RavenSystem/esp-homekit-devices #732: MagicHome RGBW shows as switch in Home app

RajeeRah, 2020-01-11:
Hi Guys,

i did managed to run the MagicHome with old Ravencore and now i changed to HAA.
the Flashing, wifi setting and adding the device in home app worked fine.

But the button in Home app looks like a switch, doesnt have a RGB selection and does not work with LEDS ... i tried this 2 version.

MagicHome RGBW LED Strip Controller (R pin 15, G pin 13, B pin 12, W pin 14):
{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"w":14,"b":[{"g":0}]}]}

and 

MagicHome RGB LED Strip Controller (R pin 15, G pin 13, B pin 12):
{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"b":[{"g":0}]}]}

Can you help me with this please?

Thanks,
Rajeeth T.

RavenSystem, 2020-01-11:
Enter in setup mode and reset HomeKit ID. Then, add again device to HomeKit.

RajeeRah, 2020-01-12:
> Enter in setup mode and reset HomeKit ID. Then, add again device to HomeKit.

Thanks for the quick answer, it did not work for me :(
I flashed 6 shelly1 with same method and they work like charm (Thanks for that).

RavenSystem, 2020-01-12:
I tested it, and it works. What HAA version are you using?

RajeeRah, 2020-01-12:
the Firmware version in Home app shows 0.2

RavenSystem, 2020-01-12:
That is the problem. Your version is too old and color lightbulbs are not supported. Update to last version (1.2.1).

RajeeRah, 2020-01-12:
> That is the problem. Your version is too old and color lightbulbs are not supported. Update to last version (1.2.1).

oh ok, can i do update it in setup mode? unfortnatley i can only change the homekit ID, json and wifi settings. Or do i have to reflash? when so, where can i get the lates main.bin ?

RajeeRah, 2020-01-12:
> oh ok, can i do update it in setup mode? unfortnatley i can only change the homekit ID, json and wifi settings. Or do i have to reflash? when so, where can i get the lates main.bin ?

cool i found it and i will try it now and let you know, thank you:
https://github.com/RavenSystem/haa/releases

RajeeRah, 2020-01-12:
Great, it works perfect, thank you very much

closed 2020-01-12 by RavenSystem