RavenSystem/esp-homekit-devices #735: Wemos D1 mini esp8266 - replica of Sonoff TH10

brunovg, 2020-01-13:
Hi there,

1rst of all, thanks to make esp8266 devices to work with Apple Homekit. This is Superb!!! :)
2nd, I have a Wemos D1 mini with a relay shield and a DS18B20 temp sensor. I had this system working with Blynk cloud, to turn on/off a water tank in my house, without any issues.
A week ago I flashed the Wemos D1 mini with HAA firmware (v1.1.2) on it, and everything was working properly on Apple Homekit, after I added some rules on EVE app to turn on/off the relay.
The problem is with the turn off rule, it seems to work only 50% of the time.
To turn on, I use a time trigger (7 am) with a temperature condition (<= 42ºC). Works!!
To turn off, I use a temperature trigger (>=45ºC) with a power condition (must be ON). Sometimes doesn't work :(
Is this a a Apple Homekit/EVE rule problem? Or maybe is something on the HAA firmware/configuration?
The Sonoff TH10/16 flawless works well with this firmware and Apple Homekit?

Thanks for your time.
Regards from Portugal.

RavenSystem, 2020-01-13:
Temperature rules fail many times. Why do you not use thermostat accessory directly?

brunovg, 2020-01-13:
> 
> 
> Temperature rules fail many times, it is not a HAA issue. Why do you not use thermostat accessory directly?

So, you're saying to use "Sonoff TH10 Heater Thermostat" instead of "Sonoff TH10 HomeKit Switch and Temperature" configuration. Am I right?

RavenSystem, 2020-01-13:
Exactly.

brunovg, 2020-01-13:
> 
> 
> Exactly.

Ok. I'll let you know if it works properly! :)

brunovg, 2020-01-15:
It work's very well with this configuration 💯 
Many thanks José 👍

closed 2020-01-15 by brunovg