RavenSystem/esp-homekit-devices #739: Unable to get shelly 1 working with HAA

MattAydin, 2020-01-14:
Hi, 

When I follow the steps stated in installation, I can't seem to connect the Shelly 1 to Homekit. 

I have attached the log file. 

Any help would be appreciated. 

With kind regards, 

Matt
[screenlog.txt](https://github.com/RavenSystem/esp-homekit-devices/files/4061618/screenlog.txt)

jsmith79, 2020-01-14:
Hey Matt, what json string did you use? There was an error in the log saying invalid json.
You should be able to get to the web interface of the Shelly by using http://HAA-B8D594.local or http://192.168.178.72

MattAydin, 2020-01-14:
I didn't use any JSON string. I am unable to access the web interface. I tried using http://HAA-B8D594.local, http://HAA-B8D594, and http://192.168.178.72. Although, when I try to navigate to http://HAA-B8D594.local, it can't resolve the address and is stuck loading. While the IP address fails completely. 

I have also read that when I do not provide a JSON string, it should enter setup mode automatically. 

> NOTE: If an invalid JSON string or no JSON string is detected during boot, the device will enter setup mode automatically.

jsmith79, 2020-01-14:
Hey Matt, which is and browser are you using? Are you able to ping the Shelly From terminal or command line?

MattAydin, 2020-01-15:
I've tried Safari and Chrome. Also, when I ping I don't get a response.

MattAydin, 2020-01-15:
So here are the steps I did to get this far. 

1. I connect my Shelly 1 with the adapter to my mac. When I connect, I make sure that GPIO0 is connected to grnd on the adapter. 
2. I erase the flash of the Shelly by executing: 
`esptool.py -p /dev/cu.usbserial-00000000 erase_flash` 
3. I disconnect and reconnect the Shelly 1 in flash mode
4. I execute: 
`esptool.py -p /dev/cu.usbserial-00000000 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m \
0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin`
5. It flashes successfully. Now, I disconnect and reconnect, not in flash mode. 
6. I take my phone and look up the created AP. (I need to disconnect and reconnect the Shelly a couple of times to successfully connect to the AP. 
7. When I am able to connect, I select my home network. Fill in the password. In the field "OTA repository:" I fill in `RavenSystem/haa`. In the field "OTA binary file:" I fill in `otaboot.bin`. I tap "Join" and this is what I receive as a log file: 

[screenlog.txt](https://github.com/RavenSystem/esp-homekit-devices/files/4064700/screenlog.txt)

RavenSystem, 2020-01-15:
Step 7 is wrong. You must fill fields as follow:

[image]

MattAydin, 2020-01-15:
Thanks for your reply. So I changed it, but now I have the same problem as stated in my first posts. I am unable to reach the device. I tried `http://haa-b8d594/`, `http://192.168.178.72/` from the network I have configured in step 7.

MattAydin, 2020-01-15:
Okay, I solved it by changing the settings in my router configuration allowing wifi devices to communicate with each other.

closed 2020-01-15 by RavenSystem