RavenSystem/esp-homekit-devices #741: Shelly 1 and PIR help

Kienz, 2020-01-15:
I try to make a normal PIR working with Shelly 1. But it doesn’t work probably. Sometimes it works and sometimes not. I think the SW input of the Shelly 1 is the problem.
I used the JSON from #651 

Any help?

[image]

vsidlu, 2020-01-20:
you can try and add "t"=0 mentioned in this post - https://github.com/RavenSystem/esp-homekit-devices/issues/632#issuecomment-573026272

this is json working for me - https://github.com/RavenSystem/esp-homekit-devices/issues/594#issuecomment-573131142

closed 2020-02-12 by Kienz

Kienz, 2020-03-20:
It worked - but not with any PIR sensor. I'm using this cheap one:
https://www.obi.de/bewegungsmelder/bewegungsmelder-180-weiss/p/4970455

Hanimaniac, 2020-03-27:
these PIR Sensors also works:
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.de%2Fulk%2Fitm%2F123703914364

lrodorigo, 2021-11-30:
> these PIR Sensors also works: https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.de%2Fulk%2Fitm%2F123703914364

Anyone else has tested this one, also with other Shellies? I'm struggling with a really similar one, bought on Amz