RavenSystem/esp-homekit-devices #742: Thermostat automation

pixidixi88, 2020-01-15:
Hi,
I have a solution for floor heating at my house based on HAA. I have one thermostat which steers a pump. It means turn on pump when water in pipes is above 30 degrees. I have also 4 valves steer by 4 HAA thermostat devices.
Is there any possibility to switch off this pump  when all of others get status “idle” close the valves? When all loops are closed pump is very noisy because water comes only through safety loop. Pump needs to be turn on again if at least one thermostat change the status to heating. Thanks for ideas.

RavenSystem, 2020-01-15:
You can do an automation with some conditions, in this case: thermostat 1 is off, th 2 is off, th 3... and turn on selected thermostat; but you need a third party HomeKit client like Home+ because stock Home App does can not do it.

pixidixi88, 2020-01-15:
Eve app cannot do this neither. Do you know which app could work? I have found many home+ apps in AppStore.

RavenSystem, 2020-01-15:
https://apps.apple.com/es/app/home-4/id995994352

RavenSystem, 2020-01-15:
You must create one automation for each thermostat to trigger it, and then include the others thermostats under conditions.

diodo157, 2020-01-15:
You can use EVE, creating rule
Must be 4.
3 which start the pump for each thermostat
and one when the condition of all three stationary thermostats causes the pump to switch off
[image]

closed 2020-01-25 by RavenSystem