RavenSystem/esp-homekit-devices #743: HomeKit Problem - Automation and Sonoff elements

honza67, 2020-01-15:
Hi, I have a problem with Sonoff elements in HomeKit. After a trouble-free use, there was a problem with automating these elements about a month ago. Switches are normally visible in HomeKit, their status is known and can be controlled (switched) from the HomeKit app. The problem is if they are included in any set up automation. E.g. at time "t" switch on or after activation of the motion sensor switch for time "t". If in this automation there is a Sonoff element together with some other element (Eve, Hue, Koogeek ...), then only this element (s) will react and Sonoff will not switch. I have several Sonnoff Basic, S26 and 2CH. The elements behave the same with both FW RavenSystem HAA 1.2.1 and FW Grupio 1.0. This seems to be a problem with uncertified elements (?). As HomeKit, I use Apple TV with the latest 13.3. Everything previously worked normally (including Sonoof) and now only HomeKit certified elements. Obviously I can't contact Apple, so I am looking for advice elsewhere .... ???

peros550, 2020-01-15:
Lately I had some issues with automation inHomekit. I was able to resolve my issues by deregistering Apple TV 4 from HomeKit and registering again.

honza67, 2020-01-16:
@peros550 
Thanks for the tip, I try this option and see .. shame that somewhere logs from Homekit are not available.

RavenSystem, 2020-01-16:
Uncertified elements have not any automation restriction. And HAA uses last Apple specifications. Many people (included me) use HAA with Apple automations, and nobody has any issue.

honza67, 2020-01-16:
@RavenSystem 
Thanks, I also used these elements for a long time without a problem and suddenly I have a problem. It's hard to find a mistake, trying what I can, but without success. But everything suggests that the problem will be on Apple TV - HomeKit center. I think the problem appeared after the last update of tvOS 13.3, but I can't say it 100%.

mafyata, 2020-01-16:
In my apple tv I have the latest tvos. Try to disable and then enable again homekit hub on your apple tv’s settings. After that apple tv must be at sleep mode, not completly turned off.

honza67, 2020-01-16:
I tried to remove Apple TV from HomeKit now, restarted and reconnected to HomeKit = the state is still the same, ie. certified elements also work in automation, Sonoof elements (no matter what FW version) work only through switching in the HomeKit :( application.

egorist, 2020-01-20:
The same problem I solved by deleting Apple ID from ATV and adding it again.

honza67, 2020-01-21:
@egorist 
Thanks for the tip. I'll try it, I'm really upset with the problem.

closed 2020-01-23 by RavenSystem