RavenSystem/esp-homekit-devices #746: MagicHome RGBW LED with external toggle connected to GPIO 0??

Waniaok, 2020-01-16:
hi, 
I tried to modify this code but to no avail, can anyone help me? in the old RavenCore this controller worked with a external toggle

> {"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"w":14,"b":[{"g":0}]}]}

https://imgur.com/a/xC2LG9J

Waniaok, 2020-01-18:
this code is almost good
>{"c":{"l":2,"b":[{"g":4,"t":5}]},"a":[{"t":30,"r":12,"g":15,"v":13,"w":14,"b":[{"g":0,"p":0},{"g":0,"t":0}]}]}

Setup Mode is on IR pads.
RGB are almost good, RGB are almost good, there are all colors but not in these places.
 any suggestions?
"r":?
"g":?
"v":?
"w":?

toshibochan, 2020-01-18:
GPIO 0 connected to FLASH button, boot fails if pulled LOW?

Waniaok, 2020-01-18:
toshibochan, you're right, it doesn't boot. why? :/

toshibochan, 2020-01-18:
GPIO0 is use for flash. So if is low on boot going on flash mode. But maybe can use as momentary push button

Waniaok, 2020-01-18:
I understand. the problem is only when connecting the power supply. push button is some idea or use other GPIO.

toshibochan, 2020-01-18:
I think sonoff basic use gpio 0 for push button but not sure. Take look.

Waniaok, 2020-01-18:
toshibochan, thank you for your help, I corrected the code:
> {"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":12,"g":5,"v":13,"w":14,"b":[{"g":4,"p":0},{"g":4,"t":0}]}]}
now Setup Mode is on GPIO 0 and external toggle works connected on IR pads :)

toshibochan, 2020-01-18:
A standard Stateless Button using GPIO0:

{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":3,"f0":[{"g":0}],"f1":[{"g":0,"t":2}],"f2":[{"g":0,"t":3}]}]}

brymonster, 2020-01-22:
Momentary ⏚ on io0 [works](https://imgur.com/a/p3WewY7) great for me.

toshibochan, 2020-01-23:
@brymonster you can share your config?

brymonster, 2020-01-23:
Try the [search!](https://github.com/RavenSystem/esp-homekit-devices/issues/696#issuecomment-569327089) 
```
{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":12,"g":5,"v":13,"w":15,"b":[{"g":0}]}]}
```

toshibochan, 2020-01-23:
i cant pair to home app. need help
[4] New 10.0.1.43:46329
[4] Pair 1/3
[5] New 10.0.1.43:46585
[5] Duplicated connection, closing old on port 46329
! [4] Disconnecting
[4] Closing connection
[5] Pair 1/3
! [5] Disconnecting
[5] Closing connection

closed 2020-02-24 by stale[bot]