RavenSystem/esp-homekit-devices #750: Unable to update

mafyata, 2020-01-17:
Hi , I have working device with Haa 0.9.1. I want to update to 1.3.1(latest) version. 
1. Enter the device in setup mode(auto ota update is on)- no luck.
2. Enter setup mode and click on “update ota” - no luck. 

I tried 2 methods many times.

mafyata, 2020-01-17:
I can update my devices from 1.2.1 to 1.3.1. The problem is only from 0.9.1

RavenSystem, 2020-01-22:
Logs, please.

Kienz, 2020-02-03:
I have the same from 1.3.1 to 1.5.x. Two devices not updating and bricked after I want to update. Both devices are not connecting to HomeKit. ping to the device is working. But I can't get logs (https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-Instructions#capture-logs).

closed 2020-03-04 by stale[bot]