RavenSystem/esp-homekit-devices #755: tywe3s boiler switch

shimon88, 2020-01-21:
Hi all!

Did someone experience with flashing HAA onto these Chinese switches with TYWE3S module (Tuya)?
here is a link:
https://www.aliexpress.com/item/32985321268.html?spm=a2g0s.9042311.0.0.15eb4c4du1O8f4

Thanks

shimon88, 2020-01-28:
So I guess no one has tried it?

tonysprenk, 2020-01-28:
I think I saw someone in the discord did a successful flash. I would be careful though because manufacturers are switching to a non ESP based chip which is not compatible with esp-HomeKit-devices.

peros550, 2020-01-28:
I found the TYWE3S chip inside a Blitwolf LT11 led strip. Had no issues flashing it with tuya convert method

tonysprenk, 2020-01-28:
Yes the tywe3s chip is esp based but there are devices being sold that have a completely new chip which is not based on esp.

shimon88, 2020-01-28:
Thanks guys. 
I have it a try but I fail to enter it to flash mode. The is a GPIO0 pad and I shorted it to ground but the esptool can’t connect to it

tonysprenk, 2020-01-28:
Does it say on the chip that it’s tywe3s?

shimon88, 2020-01-28:
Yes. 
When I short GPIO 0 and connect to usb, a red led is on. Once I disconnect the short to ground another led (green) turns on.

tonysprenk, 2020-01-28:
Maybe you can find a solution in this thread: https://github.com/arendst/Tasmota/issues/1190

shimon88, 2020-01-30:
tried that - no success.
I guess I'l leave it Tuya and home bridge for now

thanks anyway

closed 2020-01-30 by shimon88