RavenSystem/esp-homekit-devices #757: problem on boot of sonoff basic R3 cute track on garage device

pilot1981, 2020-01-21:
hi,
I have sonoff basic R3 cutted track, HAA 1.4 and close sensor on GPIO 3 (RX) on garage device with this JSON:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":3,"t":0}],"f4":[{"g":3}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}

with 1.3 and 1.4 firmware I have this problem:

When garage door is closed, Sonoff have GPIO 3 in short cut with GND; if I have a power outage, Sonoff don't boot again (it don't flash led, it don't connect to WIFI, no AP mode); to solve this I have to open contact between GPIO 3 and GND, then attach again power...it's a bug of newest firmware versions?

so to reproduce problem:
1) garage door closed = GPIO 3 in shortcut with GND
2) power OFF
3) power ON
4) sonoff don't boot

so workaround to this problem is:
1) GPIO 3 NOT in shortcut with GND
2) power OFF
3) power ON
4) sonoff boot correctly

RavenSystem, 2020-01-22:
I tested it, and I have not any issue booting device when GPIO 3 is in shortcut with GND.

pilot1981, 2020-01-22:
very strange... @RavenSystem you suppose an hardware problem?

pilot1981, 2020-01-22:
You are right: I changed hardware with new SONOFF BASIC R2 with cute track and it boot fine also when GPIO3 is in shortcut with GND.

On SONOFF BASIC R2, if I can't use GPIO 0, 2 and 15: which GPIO I can use for open sensor?

macjl, 2020-01-22:
You can Use GPIO 3 (RX)

pilot1981, 2020-01-22:
@macjl you attached a NO SENSOR to GPIO 1 Tx)? It’s in short circuit with GND?

macjl, 2020-01-22:
I've done that with some Sonoff Basic R2, mostly with contact sensors (which can be pressed at boot time).
On one of the devices, I had to add an external pull-up resistor to avoid interference, but I don't think it is related to this GPIO.

closed 2020-02-19 by RavenSystem