RavenSystem/esp-homekit-devices #758: v1.4.0 ICMP Ping - what was that??

i3laze, 2020-01-22:
As well as v1.4.0 changelog is quite descriptive, I can’t imagine a scenario how this ICMP feature might be used.

So:
You set a desired netbios hostname, and if the name (not the IP?) is pinged, HAA device toggles it’s status?

If there’s some network scanner in your LAN, you get lots of false toggles?
How about reacting then on ICMP-echo packet size length? Like 4 bytes = on, 10 = off or something?

RavenSystem, 2020-01-22:
Wiki is under development, but @WizBangCrash need more time to finish it. Please, be patient.

> 1. Looks like HAA pings some remote device IP (resolved from netbios name or FQDN) and sets own status according to remote device reachability. Right?

Yes, in the same way digital inputs work.

> 2. What is the default ping interval? 1 sec?

5 seconds, and for now there is not way to change this. But to set a ping as "not response" must be 3 fails in a row.

> 3. So there are actually two hosts: p0 and p1. How the device will react if they are different? Last event wins?

Both wins. Two hosts is like 2 buttons, what happens if you press both (one and then other)? Here is the same.

closed 2020-01-22 by i3laze

WizBangCrash, 2020-01-26:
@i3laze I've updated the Wiki with details of the new ICMP Ping feature. Let me know if any further clarification should be added to the docs.
https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#icmp-ping-inputs

i3laze, 2020-01-27:
> @i3laze I've updated the Wiki with details of the new ICMP Ping feature. Let me know if any further clarification should be added to the docs.
> https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#icmp-ping-inputs

Quite detailed. However all examples contain both reply and no-reply inputs so its hard to guess which accessory one triggers.