RavenSystem/esp-homekit-devices #762: Shutter Problem Shelly2.5

SMMMN, 2020-01-22:
Hello everybody,

I bought a shelly2.5 and successfully installed the Haa firmware. Thanks for this great work. I am already successfully switching my roller shutter with two switches. My question is now: when I press the button of the shutter in the home app it only switches between fully open and fully closed. Is it possible that the roller shutter stops when I press the button in the home app a second time?

marcoscv-work, 2020-01-23:
You can use the percent opening/closing option by using long-press in your roller

[image]

SMMMN, 2020-01-23:
Thanks, I found that out too. So is this the normal HomeKit behaving, like a lamp that is either on or off, and so the blinds are fully open or fully closed? It would actually be easier for me to stop the blind with a second tap on the button in the Home app.

RavenSystem, 2020-01-25:
@SMMMN Ask Apple to implement stop function in HAP.

marcoscv-work, 2020-01-26:
I see your point...  yeah, sometimes I have to run to press the switch :-P  or you can order to siri to stop in an opening percent...

closed 2020-01-31 by RavenSystem