RavenSystem/esp-homekit-devices #763: RavenCore current state

i3laze, 2020-01-23:
I still find HAA too complicated for generic use. Too advanced it is.

Huge flexibility comes at multiple disadvantages like:
- You need to know device IP address to set it up.
- You need to read JSON and don’t mess up with the {[{markup}]}.
- No startup templates - you get one from wiki and hope it will work.
- Entering setup is more complicated. 

I’ve just upgraded f/w on RC v1.2 and eventually got HAA (I hoped to get RC v1.3.1).
then I erased flash and configured LCM with ”RavenSystem/ravencore” repo, but again I received HAA!
Is that by design now? I was totally OK with RC and an ability to move forward to HAA via OTA when I decide.

RavenSystem, 2020-01-23:
> You need to know device IP address to set it up.

You can use hostname instead (Showed in Home App under Serial number label).

> You need to read JSON and don’t mess up with the {[{markup}]}.

Use a JSON validator. Many people use JSON to configure services like HomeBridge without any issue.

> No startup templates - you get one from Wiki and hope it will just work.

All templates from the Wiki page work perfectly.

> Entering setup is more complicated than it was via Eve.app.

Entering setup is very easy. Even you can define custom ways to do that.

> Any changes require significant time and hustle.

No, if you know what are you doing.

RavenSystem, 2020-01-23:
Read the Wiki and release notes. You can get info about RavenCore here:
https://github.com/RavenSystem/esp-homekit-devices/releases/tag/RC_1.3.0
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore

i3laze, 2020-01-23:
In the end, the current only option to install RC is without OTA?

RavenSystem, 2020-01-25:
Yes, but RC is not currently available.

closed 2020-01-25 by i3laze