RavenSystem/esp-homekit-devices #764: fail to pair

toshibochan, 2020-01-23:
fail to pair to home app 

[4] New 10.0.1.43:46329
[4] Pair 1/3
[5] New 10.0.1.43:46585
[5] Duplicated connection, closing old on port 46329
! [4] Disconnecting
[4] Closing connection
[5] Pair 1/3
! [5] Disconnecting
[5] Closing connection

danielssantos, 2020-01-25:
Same problem here. 3 devices, none are found by apple home. It only works if you do the flash without the OTA. Trying to access by HAA-XXXXXX did not work, but by the ip, yes. I put the JSON tag and nothing advances after that. 2 devices worked with RavenCore perfectly, the other is new. I was unable to capture logs. I have active devices running RavenCore.

RavenSystem, 2020-02-19:
Complete logs are needed.

closed 2020-02-19 by RavenSystem