RavenSystem/esp-homekit-devices #767: Salt level sensor for water softener

pilot1981, 2020-01-26:
Hi,
There is a chance to have new device Salt level sensor for water softener? This is already allowed by Apple?

I need only a sensor 0/1 to to when to add new salt in my water softner...

In worst case I could use a sonoff/Shelly with a sensor attached to a micro switch which go on/off depending from salt level...

RavenSystem, 2020-02-19:
HomeKit does not provide a salt level sensor. You can use any other supported sensor (Maybe filter change sensor could be a good option): https://github.com/RavenSystem/esp-homekit-devices/wiki/Sensors

closed 2020-02-19 by RavenSystem

pilot1981, 2022-12-11:
I used a JSN SR04T with water leak sensor HAA accessory and a fixed distance: it works fine!