RavenSystem/esp-homekit-devices #768: HAA_boot.bin release

i3laze, 2020-01-27:
Great job on the new one.

Looks like a fork of LCM. 
Any benefits from reflashing existing LCM+HAA devices to this OTA?

macjl, 2020-02-04:
Hello,

Is there a way to flash the new OTA firware (with Serial), and keep HAA all settings (JSON, Homekit data, etc...)?

I would like to keep all my automations and settings in Home

RavenSystem, 2020-02-04:
@macjl Yes (I think), but try first with a device without automations to have not a big lost.

- Save a copy of your JSON.
- Flash it with this command, using only haaboot.bin (Don't erase it before):
`esptool.py -p /dev/<your_ESPPort> --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x2000 haaboot.bin`
- Connect to it, and configure WiFi and JSON (Don't reset HomeKit ID).
- Click Save, and wait about 5-7 minutes.

RavenSystem, 2020-02-04:
@i3laze 

> Any other changes unnoticed?

Removed the Private Key to verify LCM firmware.

> Any benefits from reflashing existing LCM+HAA devices to this OTA?

A bigger user storage to save greater JSON configs.

> Any way to reflash boot partition without Serial?

No.

> What other features are on the roadmap?

None.

closed 2020-02-04 by i3laze

macjl, 2020-02-07:
> Yes (I think), but try first with a device without automations to have not a big lost.
>
> * Save a copy of your JSON.
> * Flash it with this command, using only haaboot.bin (Don't erase it before): [...]

I tried it on a tes device, and it worked!
I had to reboot twice after last step, but all works perfectly now! Thank you!

peros550, 2020-06-02:
> @macjl Yes (I think), but try first with a device without automations to have not a big lost.
>
>
>
> - Save a copy of your JSON.
>
> - Flash it with this command, using only haaboot.bin (Don't erase it [...]

Just for me to understand. Are those steps used to go from LCM+HAA —> HAA_OTA + HAA preserving existing HomeKit ID?

i3laze, 2020-06-02:
> Just for me to understand. Are those steps used to go from LCM+HAA —> HAA_OTA + HAA preserving existing HomeKit ID?

Yeap, you’ll get up-to-date from any HAA revision.

peros550, 2020-06-02:
So this is a way to go from LCM to HAA_ OTA right ?