RavenSystem/esp-homekit-devices #771: Garage door opener - No response 50% of the time

lockpicker, 2020-01-29:
I've been using a (homebridge + wemo maker) garage door controller on HomeKit for over 18 months without any issues. 
I decided to switch to a native HomeKit garage controller with a HAA on Sonoff Basic setup in the last few weeks. 
HAA on son off basic as a garage door controller seems to be unresponsive rather frequently (about 50% of the time or more). 
Just to isolate any hardware issues, I flashed another Sonoff basic with HAA which shows the same issue, to isolate any network issues (although given all my other HomeKit devices work perfectly, I don't think this is a problem), I tried it at a friends place (his network) and the setup displayed the same issues.

This is causing significant reliability issues as I can't trust the setup to work every time. 
Any pointers would be much appreciated. 

[image]

ymyshko, 2020-01-29:
I would advise checking the signal level at the module installation point. mDNS does not forgive the loss of packets and long pings. If signal level is -70db or less, there will be similar problems. Homebridge server can stand closer to the Wi-Fi access point and you just don’t see the inaccessibility of the module to homekit.
Sorry for my english&

wwns, 2020-01-29:
Check what is happening, connect the module to the Uart port, if you have a Mac computer:
ls /dev/tty.*
screen /dev/tty.xxx_xxxx 115200
windows putty 👍
you'll know what's wrong!

lockpicker, 2020-01-30:
Thanks @ymyshko and @wwns 
I’ll give these a shot and report back.

Huege, 2020-02-11:
Dear all,
I'm facing the same issue.
log/console did not give any hint about an problem. Signal strength is ok.
Is it only a garadoor or a general issue?
BR

yusufd1, 2020-02-14:
A lot of no response issues are now resolved in 1.6.4. Also wifi signal at the device must be pretty strong probably minimum of -75db. Check the signal strength and ensure that your have at least around that amount of strength.

closed 2020-03-15 by stale[bot]