RavenSystem/esp-homekit-devices #773: Smoke Sensor problem - Sonoff Basic (new board R2)

Waniaok, 2020-01-29:
hi, 
this is my code for smoke sensor (GIPO3-RX) with external toggle (GIPO2)

> {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":2},{"g":2,"t":0}]},{"t":8,"i":10,"f1":[{"g":3}]}]}

The problem is that when GIPO3 connects to GND, nothing happens, but when I disconnect GND, a smoke detection message appears. works like a push button. smoke detector is MQ-2

wwns, 2020-01-30:
it's am analog sensor at 5V. esp8266/5 suports only 3V3

Waniaok, 2020-01-30:
this sensor have digital output and analog too, but not the sensor is problem

Waniaok, 2020-01-30:
I solved my problem:
> {"c":{ "l": 13},"a":[{"t":1,"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":2},{"g":2,"t":0}]},{"t":8,"f1":[{"g":3,"t":0}],"f0":[{"g":0,"t":1}]}]}
thanks to this code I have a cheap smoke detector

closed 2020-02-03 by RavenSystem

EgorMuhin, 2021-04-15:
> I solved my problem:
> 
> > {"c":{ "l": 13},"a":[{"t":1,"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":2},{"g":2,"t":0}]},{"t":8,"f1":[{"g":3,"t":0}],"f0":[{"g":0,"t":1}]}]}
> 
> thanks to this code I have a cheap smoke detector

Hello, please tell me what is the GPIO connection here and how did you build the MQ-2 sensor itself, how much is it sensitive?