RavenSystem/esp-homekit-devices #774: Xiaomi-Philips bulb support

MaxLesichniy, 2020-01-29:
Hi, I really liked the project. Already flashed 5 devices on haa. But I wanted to reflash the Xiomi Philips led bulb. Appearance, she uses a different principle to control color temperature and intensity. This is described here https://github.com/arendst/Tasmota/issues/3055.

Can HAA support this type of LED control?

dbartelmus, 2020-02-03:
So did you open your bulb already? Can you show us internals?

PWM is OK, DIgital not as far as I asked for RGBW strip before.

MaxLesichniy, 2020-02-05:
Here is a photo of the board. The driver microcircuit marking is jammed, the lamp is in working condition.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

dbartelmus, 2020-02-07:
As far as I know PWM has separate channels (wires) for R G and B, you have combined so it's digital communication (1 lane). HAA is not supporting it yet, ask politely author about this.

dbartelmus, 2020-02-07:
Wait, this is not RGB bulb? Fool me.

Go ahead, plug RX, TX GND and V and try to flash it... simple. Trace signal wire with multimeter to find out which GPIO is it, looks like IO15 or IO12.
I would rather change JSON few time to figure this, in worst case scenario you have 15 tries

MaxLesichniy, 2020-02-11:
@dbartelmus  Thanks for answers. 
Thanks for answers. 
> Wait, this is not RGB bulb? Fool me.

No, this is not an RGB bulb. This is a white bulb with adjustable light temperature.

I quote:
> To make it short I discovered GPIO12 controls color temperature without modifying total intensity, and GPIO15 controls light intensity without changing color temperature. It is different from other lamps where one GPIO usually controls the cold leds, and another one the warm leds, so I had to modify the code to accomodate this lamp.

GPIO12 controls the temperature by PWM. Warm and cold channel connected inverted to each other. When the PWM signal is at a high level that cold LEDs turn on, and when at a low level - warm LEDs turn on. When PMW operating cycle is 100% - the bulb shines completely cold light. At ~ 0%  - bulb shines warm light.

GPIO15 - is the same signal, but this pin is adjusts the overall brightness of all LEDs.

I am ran the light bulb with follow configuration:

```
{
  "c": {
    "b": [
      {
        "g": 0
      }
    ]
  },
  "a": [
    {
      "t": 30,
      "w": 15,
      "s": 0,
      "p": 1024,
      "d": 1,
      "e": 10,
      "b": [],
      "fg": 1,
      "fr": 1,
      "fv": 1,
      "fw": 1
    },
    {
      "t": 30,
      "s": 0,
      "p": 1024,
      "d": 1,
      "e": 10,
      "fr": 1,
      "fg": 1,
      "fv": 1,
      "b": [],
      "w": 12,
      "fw": 1
    }
  ]
}
```
The Home application has two sliders (as two accessories) now. One slider are control a light temperature, and the other - controls the intensity.

Can the firmware author add support for such bulb?
I don't know the code as the author knows it :/
I suggest add a new parameter to json file, which will indicate the temperature control pin. For example:
```
....
"a": [
    {
      "t": 30,
      "w": 15,
      "tm": 12, <--
      "s": 0,
      "p": 1024,
      "d": 1,
      "e": 10,
      "b": [],
      "fg": 1,
      "fr": 1,
      "fv": 1,
      "fw": 1
    },
 ...
```

dbartelmus, 2020-02-11:
Can you show some screenshots from your accessory inside home app?
One slider for brightness and color wheel for light temperature sounds right
[image]

MaxLesichniy, 2020-02-11:
I don't have color wheel for light temperature. I have two sliders. And this sliders as a two accessories.

[image]

dbartelmus, 2020-02-11:
Privet! Please join discord channel to get more help, I do not have so much experience in this area but it can be done proper somehow for sure.
https://discord.gg/v8hyxj2

RavenSystem, 2020-02-19:
Sorry, I have not any future plan to support this kind of devices.

closed 2020-02-19 by RavenSystem