RavenSystem/esp-homekit-devices #777: lock key button to esp

vsidlu, 2020-01-31:
[image]

{"c":{"o":1,"l":2,"i":1,"b":[{"g":0,"p":1,"t":1}]},"a":[{"t":4,"v":1,"f0":[{"g":14,"v":1}],"f1":[{"g":14,"v":1}],"0":{"r":[{"g":14,"v":0,"i":2}]},"1":{"r":[{"g":14,"v":0,"i":1}]}}]}

need help with json -

lock opened when button is pushed twice. locks closed when button is pushed once. how can i make two digital output changes in a row so the lock is opened? with json above it only closes.

vsidlu, 2020-02-06:
@RavenSystem is there a way to imitate double press? starting to get crazy sorting it out! can you help me?

RavenSystem, 2020-02-07:
I am not sure what you want. It is?

Button on GPIO 14:
- Double press opens lock.
- Single press closes lock.

`{"c":{"o":1,"l":2,"i":1,"b":[{"g":0}]},"a":[{"t":4,"f0":[{"g":14,"t":2}],"f1":[{"g":14}],"0":{"r":[{"g":14,"i":2}]},"1":{"r":[{"g":14,"i":1}]}}]}`

(I removed all default and unused keys).

vsidlu, 2020-02-07:
I want when I press 
- open lock in homekit - gpio 14 is changing from high to low to high to low and back to high.
- close lock in homekit - gpio 14 is changing from high to low and back to high.

1.- I’ve noticed that the GPIO needs to be HIGH STATE as normal, and when you want to activate it, you need to write LOW_STATE in the GPIO.
2.- 1 pulse from high to low close the lock, 2 pulses from high to low open it…
3.- If I put HIGH STATE as normal mode since Boot, there’s a glitch that tries to close the door when the ESP8286 boots

RavenSystem, 2020-02-07:
Use my JSON.

> 1.- I’ve noticed that the GPIO needs to be HIGH STATE as normal, and when you want to activate it, you need to write LOW_STATE in the GPIO.

You can change this adding `"i":1` in the first GPIO 14 declaration.

> 3.- If I put HIGH STATE as normal mode since Boot, there’s a glitch that tries to close the door when the ESP8286 boots

When device boots, it will execute actions of actual lock state, in this case, action `"1"`, related to closed state.

vsidlu, 2020-02-07:
i checked your JSON, but it is not what I want.

I need digital output 1 -> 0 -> 1  -> 0 so it imitates two button pushes.

digital output on gpio
1 pulse from high to low close the lock, 
2 pulses from high to low open it…

RavenSystem, 2020-02-07:
Ahh, now I understand what you want.

`{"c":{"o":1,"l":2,"b":[{"g":0}]},"a":[{"t":4,"0":{"r":[{"g":14,"v":1,"i":1},{"g":14,"i":1.5},{"g":14,"i":0.5}]},"1":{"r":[{"g":14,"i":0.5}]}}]}`

vsidlu, 2020-02-07:
Thanks! Needed small modification in inching time but now it works - 
{"c":{"o":1,"l":2,"b":[{"g":0}]},"a":[{"t":4,"0":{"r":[{"g":14,"i":0.5},{"g":14,"v":1,"i":0.6},{"g":14,"i":2}]},"1":{"r":[{"g":14,"i":0.5}]}}]}

Thank you for your effort! Now I am heading to search donation button! Have a nice weekend!

closed 2020-02-07 by vsidlu

RavenSystem, 2020-02-07:
Many thanks!!