RavenSystem/esp-homekit-devices #780: Again problem with close sensor of garage device

pilot1981, 2020-02-02:
Hi,
I use SONOFF BASIC R2 with cutted track for my garage door with this JSON:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":3,"t":0}],"f4":[{"g":3}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}

For close sensor I use magnetic sensor on GPIO3.

In the last two months I had 2 different hardware with a same issue: GPIO3 don’t work fine so close sensor don’t work properly...

The behaviour is always the same: new hardware, all work fine for some weeks and then close sensor don’t work so the garage door remain always in opened status in Apple home app (in real world garage door open and close many times in the day).

It’s a common bug of SONOFF BASIC R2? It’s related to HAA or I have to use some type of resistor on GPIO3?

macjl, 2020-02-02:
I had similar issue with one of my switch. I think the wire were too long, and get interference from high voltage. I burned 3 sonoff devices at this place. Then I added a pull-up resistor of 47k between +3,3v and GPIO, and all working fine till then.

pilot1981, 2020-02-02:
@macjl resistor of 47k between +3,3v and GPIO3?

pilot1981, 2020-02-03:
@macjl I can do it on defective sonoff or I have to use a new HW? Old is already damaged?

macjl, 2020-02-03:
You can try. For me, th interferences totally burned 1 of my sonoff. But an other one worked again after adding the pull-up.

pilot1981, 2020-02-04:
[image]

So, I have also to change JSON to invert close sensor from NC to NO?

macjl, 2020-02-04:
Yes! And you can also disable internal pull-up reistor if not already done.

pilot1981, 2020-02-04:
@macjl how I can disable internal pull-up resistor? I have to do it in JSON?

macjl, 2020-02-04:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#internal-pull-up-resistor

pilot1981, 2020-02-04:
ok thanks, so I have to add 

"p" | 0 | No pull-up resistor required
-- | -- | --

and revert close sensor to my JSON:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":3,"t":0}],"f4":[{"g":3}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}

any help?

macjl, 2020-02-04:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":3,"t":1,"p":0}],"f4":[{"g":3,"t":0}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}

pilot1981, 2020-02-04:
thanks @macjl but I don't understood all changes you proposed...

"p":0 OK to disable internal pull-up resistor

but why change to "t":1 and "t":0 ?

to revert close sensor from NC to NO I have to change "i" accessory?

https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#inverted-digital-input

I don't understand...

macjl, 2020-02-04:
Both works.
t : 1 => triggered when GPIO : fom GND to 3.3v
t : 0 => triggered when GPIO : from 3.3v to GND

I don't know what is the best solution : 
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":3,"t":0,"p":0,"i":1}],"f4":[{"g":3,"i":1}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}
or
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":3,"p":0}],"f4":[{"g":3,"t":0}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}

I choosed shorter JSON

RavenSystem, 2020-02-04:
Both options do exactly the same, but short JSON uses less RAM memory.

pilot1981, 2020-02-04:
@macjl @RavenSystem thanks very much for your answer! This afternoon I go to test pull-up resistor and JSON.

I would like to understand better to become independent in the JSON creation:

why in short version I don't see "i" option? from wiki I see "i":0 normal input is the default, but I need "i":1 as here in function F3 when garage door is closed:

"f3":[{"g":3,"t":0,"p":0,"i":1}

so I expect to have always "i":1 declared because it in't default value...
and I when you declare "i":1, you add also "t":0

in function F4, when garage door is opening, I see you replaced "i":1 (not daft) with "t":0...I don't understand...

I think I have some problem to understand howto use "T" value; in wiki I see this:

"the button is activated by a single press ("t": 1)"

so I understand you used "t" value to create the result of action to press a "virtual" button that: 

A) in function F3 when garage door is closed (close sensor = 1 because GPIO3 shortcircuited with GND) you have:
t : 0 => triggered when GPIO3 : from 3.3v to GND
it mean close sensor go to low signal 0?

B) in function F4, when garage door is opening (close sensor = 0 because GPIO3 NOT shortcircuited with GND), you don't declare "t": 1 because it is default value???
t : 1 => triggered when GPIO3 : from GND to 3.3v
it mean close sensor go to high signal 1?

closed 2020-02-05 by pilot1981

macjl, 2020-02-06:
Try to explain it with a schema : 
[image]

pilot1981, 2020-02-06:
thanks very much @macjl: result is very clear for me!

but I don't understood the use of accessory "t" and "i"...