RavenSystem/esp-homekit-devices #782: Sonoff basic light switch and contact sensor, drops out.

xtremedays, 2020-02-06:
Hi all

I have this issue with a sonoff basic, programmed to be a light switch with a toggle switch (gpio1) and a contact sensor on gpio3.

It works sometimes but most of the time is 'no response'. I can still use the toggle switch to turn the light on and off normally. 

I had it working flawlessly until I had to reset HomeKit. After adding all the accessories back in (about 20 sonoffs), this is only the one which has issues. I have another sonoff basic (as a light switch) right next to it and have no issues, so not a power or wifi range issue. I have tried another sonoff basic and have the same issue. 

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":1},{"g": 1,"t": 0}]},{"t":5,"f0":[{"g":3,"t":0}],"f1":[{"g":3}]}]}

Any thoughts?

On a separate issue, ive been having issues with remote access since the reset too... Apple TV 4 is my hub.

TIA

tonysprenk, 2020-02-06:
What doe your network look like? I was having an issue where the device was connected to specific AP and would only be controllable if my iPhone was also connected to that AP. This would also mean that I wouldn’t be able to access outside the network as the device was not connected to the same AP as my home hub.

xtremedays, 2020-02-06:
Only one router. All devices are connected to it.

xtremedays, 2020-02-07:
Hardware issue. The sonoff doesn't have very good range for some reason. Tried it with another device and its all good...

closed 2020-02-07 by xtremedays