RavenSystem/esp-homekit-devices #783: Accessory already added

DanielMosco, 2020-02-06:
Hello. Due to wrong color pins in Magic Home RGBW example i had to reflash device as it didn't enter setup mode. Now when i try to readd accessory after flashing, Home app says that accessory already added. Is there a way to readd it without erasing the entire home setup???
Thanks in advance, 
Daniel

mafyata, 2020-02-06:
Enter setup mode and reset id. Why did you reflash? You can change pins (json) in setup mode.

DanielMosco, 2020-02-06:
I have reset ID several times. I changed pins after reflashing. I was unable to enter setup mode without reflashing. I switched 8 times in home app, i see HAA-XXXX device but unable to connect. Now, after scanning QR code, i'm getting the "already added" even while the device isn't connected to power.

mafyata, 2020-02-06:
Try to restart phone and hub.

DanielMosco, 2020-02-06:
Restarted modem + router, restarted phone, restarted Apple TV. Restarted even the refregerator :)
Nothing. As soon as i scan the QR code, i'm getting the same message.

pilot1981, 2020-02-06:
Try also normal code...

DanielMosco, 2020-02-06:
Tried already. It just keeps searching for device and no luck. I checked that device is in my network.

RavenSystem, 2020-02-06:
Enter in setup mode, reset homekit ID, and then add it using option “I have not a code or I can not scan it” from Home App.

DanielMosco, 2020-02-06:
Tried it already. Same shit. 
Is there a way to enter setup on magic home device? I have no buttons on it. I did the setup by reflashing it every time.

RavenSystem, 2020-02-06:
Did you erase firmware each time you reflashed it?

DanielMosco, 2020-02-06:
> Did you erase firmware each time you reflashed it?

Yes!!!

RavenSystem, 2020-02-06:
> Is it still possible to flash with old OTA Ravencore? I have 1.2 locally on my PC.

No.

You can use emergency setup mode. It is always available.

What JSON are you using?

peros550, 2020-02-07:
I had a similar situation in a custom HomeKit firmware. No matter what, during pairing the device was rejected with the message : already paired. 

Finally I was able to add it but first I had to change the mDNS name and the HomeKit code to something different than previous state.

DanielMosco, 2020-02-07:
> > Is it still possible to flash with old OTA Ravencore? I have 1.2 locally on my PC.
> 
> No.
> 
> You can use emergency setup mode. It is always available.
> 
> What JSON are you using?

I’m using Magic Home RGBW from the examples page. I just modified RGB pin numbers.

DanielMosco, 2020-02-07:
As i ment

> I had a similar situation in a custom HomeKit firmware. No matter what, during pairing the device was rejected with the message : already paired.
> 
> Finally I was able to add it but first I had to change the mDNS name and the HomeKit code to something different than previous state.

I don't understand nothing from what solved your problem. Sorry...

DanielMosco, 2020-02-07:
As i mentioned before, I don't even have a device powered on. AS soon as i choose "Add accessory" and then scan the QR code provided here, I'm getting the "already added" message. This means, as i understand, that next time i will try to add another device with RavenCore software on board and i will scan the QR code, i will get this message again.

DanielMosco, 2020-02-07:
Just done something different. My friend has 1 device using the same software. I just chose "add new device" and scanned the same QR code. Same error: "Accessory already added". Different iphone, different home. The only common thing in these setups is: same FTDA adapter and same MacBook performed flashing. 
Any ideas???

RavenSystem, 2020-02-07:
If you do same steps with a similar hardware, result will be the same. 

Send logs, please.

RavenSystem, 2020-02-07:
And read Wiki about loading setup mode web page. Wiki is here to be read, and avoid to generate false issues by users.

DanielMosco, 2020-02-07:
> If you do same steps with a similar hardware, result will be the same.
> Send logs, please.

> And read Wiki about loading setup mode web page. Wiki is here to be read, and avoid to generate false issues by users.

I red wiki and was able to call setup mode. I'm getting "unable to connect to HAA-XXXXXX"  while trying to connect to HAA-XXXXXX. So I'm erasing and reflashing to be able to get the setup page. 
It is not a false issue. This is the first time since ever i'm having any issue with RavenCore software and i flashed more than 20 (Shelly's, Sonoffs, MagicHomes). My house entirely operates with the help of your firmware. 

1 thing I cannot understand is how someone else with different iPhone and different home setup is trying to add accessory and after scanning the same QR code, getting this "already added" error without even connecting no device, just scanning the barcode.

RavenSystem, 2020-02-07:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#device-connection

DanielMosco, 2020-02-07:
Ravencore, sorry man but you don't read the main issue. I know how to setup and i red it from start to end and vice versa. There is no need to copy/paste different pages. 

> "1 thing I cannot understand is how someone else (actually 2 people) with different iPhones and different home setups is trying to add accessory and after scanning the same QR code, getting this "already added" error without even connecting no device, just scanning the barcode."

Thanks anyway. You can close this unsolved issue. I will patiently wait until someone else will have it. 
Have a nice day, man.

macjl, 2020-02-07:
Don't you have an other HAA device on your network with same MAC Address end?
HAA keeps last 6 digits of MAC Address for identification. So there is a little chance to find devices that have the same.

DanielMosco, 2020-02-07:
> Don't you have an other HAA device on your network with same MAC Address end?
> HAA keeps last 6 digits of MAC Address for identification. So there is a little chance to find devices that have the same.

Please read my last own quote. I don't mention any device at all at this stage. 
Thanks anyway.

peros550, 2020-02-07:
Daniel, 

Are you able to provide logs of the moment when you are trying to add your device to home app?

DanielMosco, 2020-02-07:
> Daniel,
> 
> Are you able to provide serial capture (logs) of the moment when you are trying to add your device to home app?

Sorry, but how i can capture any log if i just open the home app, choose "+", choose "add accessory", then scanning the QR code and finally getting the "already added". Flashed device is sitting aside UNPLUGGED, NO POWER. 
Same situation is with my 2 friends, each one with his own home setup.

tonysprenk, 2020-02-07:
I think that the message will always display unless there is a new “HomeKit ID” on the network. So if you are connected to a network that has already some HAA devices connected or has had HAA devices connected at any point it will always show the message unless a new HomeKit ID is added to the network. I would advise to flash again, then go into setup mode, then choose reset HomeKit ID, then wait a few minutes, then try to scan the wrong code again. Also make sure the device and the phone are connected to the same AP. Please correct me if I’m wrong.

DanielMosco, 2020-02-07:
> I think that the message will always display unless there is a new “HomeKit ID” on the network. So if you are connected to a network that has already some HAA devices connected or has had HAA devices connected at any point it will always show the message unless a new HomeKit ID is added to the network. I would advise to flash again, then go into setup mode, then choose reset HomeKit ID, then wait a few minutes, then try to scan the code again. Also make sure the device and the phone are connected to the same AP. Please correct me if I’m wrong.

Already done Homekit id reset. No changes. Thanks

tonysprenk, 2020-02-07:
> Already done Homekit id reset. No changes. Thanks

Ok, maybe a hardware issue then. When you reset ID of a different device and add it again in the home app does it work?

peros550, 2020-02-07:
> Sorry, but how i can capture any log if i just open the home app, choose "+", choose "add accessory", then scanning the QR code and finally getting the "already added". Flashed device is sitting [...]

What setup are you using to flash? Could you please describe?

RavenSystem, 2020-02-07:
IMO device can not be added because it never runs HomeKit server and does not announce mDNS. Message is related to other device in the same network that is already paired. I ask for JSON used (a screenshot of setup web page could be a good point) and logs, but nothing.

DanielMosco, 2020-02-07:
{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,”r”:12”g”:5”v”:13,”w”:15,”b”:[{“g”:0}]}]}

RavenSystem, 2020-02-07:
`{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,”r”:12”g”:5”v”:13,”w”:15,”b”:[{“g”:0}]}]}`

It is a malformed JSON. Some quotes are wrong:
`"` -> Right.
`”` -> Bad.

tonysprenk, 2020-02-07:
Is it right that when the json is not valid the device will go automatically in setup mode? And in setup mode you cannot add it to home app?

RavenSystem, 2020-02-07:
> Is it right that when the json is not valid the device will go automatically in setup mode?

Yes.

> And in setup mode you cannot add it to home app?

Yes, because without a valid JSON, device does not know what HomeKit accessory must be. And without HomeKit server running, there is not mDNS announce and subsystem to pair is down. And when you try to pair it with QR code, an error message appears telling about other device in the same network that is already paired.

tonysprenk, 2020-02-07:
@RavenSystem that makes a lot of sense! Thanks for the info!

DanielMosco, 2020-02-07:
> `{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,”r”:12”g”:5”v”:13,”w”:15,”b”:[{“g”:0}]}]}`
> 
> It is a malformed JSON. Some quotes are wrong:
> `"` -> Right.
> `”` -> Bad.

Hey and thanks very much for your and everybody's help. Problem was solved. 
1st of all, this JSON was copy/pasted from "examples" page in wiki, so currently wrong quotes are in that example. Possibly need to change. But this code works well, but with incorrect colors (in my device)
2nd, After copying the JSON string, i've edited the JSON, but accidentally erased "comma"s between the r,g,v,w.
You can see i the string i've posted. 
After editing the JSON with commas, everything worked as magic. 
As i understand, with wrong JSON string device should've go to setup mode, but it didn't happen. 
Anyway, this was the only trouble shooting i've ever had. I'm new to HAA and old to RC+Eve. 
Thanks to all!!!

RavenSystem, 2020-02-07:
> 1st of all, this JSON was copy/pasted from "examples" page in wiki, so currently wrong quotes are in that example.

Sure?
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#magichome-rgbw-led-strip-controller-r-pin-15-g-pin-13-b-pin-12-w-pin-14

DanielMosco, 2020-02-07:
> > 1st of all, this JSON was copy/pasted from "examples" page in wiki, so currently wrong quotes are in that example.
> 
> Sure?
> https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#magichome-rgbw-led-strip-controller-r-pin-15-g-pin-13-b-pin-12-w-pin-14

I'm sure what was 10 minutes ago. I've copy/pasted from the same place you refer. It is strange. I still have the copy/pasted string with wrong quote marks.

RavenSystem, 2020-02-07:
> I'm sure what was 10 minutes ago. I've copy/pasted from the same place you refer. It is strange. I still have the copy/pasted string with wrong quote marks.

If you edit JSON string with an iPhone/iPad after copy and paste it, quotes will be change due to orthographic corrector. Check it.

closed 2020-02-07 by RavenSystem

DanielMosco, 2020-02-07:
> > I'm sure what was 10 minutes ago. I've copy/pasted from the same place you refer. It is strange. I still have the copy/pasted string with wrong quote marks.
> 
> If you edit JSON string with an iPhone/iPad after copy and paste it, quotes will be change due to orthographic corrector. Check it.

I've edited with macbook. Possibly the same. 
Thanks

morphix9, 2020-02-07:
Having the same problem with the latest HAA and OTA version. This is my 10th Shelly I'm flashing / adding. I was using HAA 1.2 before and old OTA and every worked fine. And now I'm getting that the device has been already added (second). 

My JSON is the same as for many other devices:
{"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}],"s":0}]}

What's the easiest way to get some logs down? I'm sorry, but the wiki is not really clear to me.

lpbas, 2022-08-03:
I've had the same issue as well, but my mistake was that I was not on the correct WiFi network on my phone. There goes 45 minutes of debugging 😅

So to recap what has been said in this issue, if you're getting the accessory already added it means that HomeKit cannot find the HAA's HomeKit Server running due to one of the following reasons:

1. You've inputted an invalid configuration JSON, as explained [here](https://github.com/RavenSystem/esp-homekit-devices/issues/783#issuecomment-583355446) and your device is in Setup Mode. When in Setup mode the HomeKit Server is *not* [running](https://github.com/RavenSystem/esp-homekit-devices/issues/783#issuecomment-583357522), and you can't add the device to HomeKit.
2. You are not connected to the same WiFi network as your HAA device. Double check your phone's WiFi!