RavenSystem/esp-homekit-devices #785: Sonoff Basic R2 with DHT22 as thermostat

teymur1988, 2020-02-07:
Hi @RavenSystem,

First, thanks a lot for the HAA! Great, just great thing! Could you please advise:

I've got a sonoff basic with DHT 22 attached to it (using GPIO 2) and here is my config. I use Heater & Cooler mode, because the goal is to keep the room temperature "in between" say 22-24 degrees. All works no problem, but somehow after some random time it stops working, not updating the sensor data, sonoff led flashes 3 times every 1 second, and turns the output off. Reconnecting the sensor brings it alive for some time. Sensor wiring seems to be alright, I have a 10 kOhm resistor between 3.3v and data ping on the sensor. What could be wrong? Here is my config:

`{"c":{"o":0,"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":0}]},"2":{"r":[]},"3":{"r":[{"g":12,"v":1}]},"4":{"r":[]},"5":{"r":[]},"t":21,"b":[],"g":2,"n":2,"w":3,"m":20,"x":30,"s":4,"f3":[],"f4":[],"f0":[{"g":0,"t":0}],"f1":[{"g":0,"t":1}],"j":3}]}`

Regards

Teymur

RavenSystem, 2020-02-07:
Maybe your sensor needs more time to rearm. Try increasing polling period j from 3 to 5.

teymur1988, 2020-02-07:
> Maybe your sensor needs more time to rearm. Try increasing polling period j from 3 to 5.

Thanks you, I will try that and see how it goes and report back. 

Do you think there could be an issue if using "Heater & Cooler" mode? Because when it has reached the target temperature and starts to cool there is no output defined.

Regards

Teymur

RavenSystem, 2020-02-07:
> Do you think there could be an issue if using "Heater & Cooler" mode?

No.

Some sensors need more time to wait between reads. It seems that your sensor needs 3 seconds, but if firmware is doing other task at the read moment, it will take read at 3.1 seconds instead 3.0, but next time will be at 2.9, and in that moment, your sensor will not response anymore until you reset sensor unplugging and plugging it again.

teymur1988, 2020-02-07:
> No.
>
> Some sensors need more time to wait between reads. It seems that your sensor needs 3 seconds, but if firmware is doing other task at the read moment, it will take read at 3.1 seconds instead [...]

Thanks. I've set the polling period to 5 now, and still testing. So far so good. Hasn't dropped. Let's see how it goes. Could you please keep the issue open for now, it that's OK with you? I hope it doesn't drop at all and I'll reply and we'll close the issue. Thanks.

Regards

Teymur

RavenSystem, 2020-02-07:
Could you please keep the issue open for now, if that's OK with you?

Sure.

teymur1988, 2020-02-09:
Hi @RavenSystem 

This seems to be working fine, I’ve not seen any drops since I changed the polling interval to 5. Thanks a lot. Please set to resolved.

Regards

Teymur

teymur1988, 2020-02-10:
Update:
It dropped today again after being stable for 3 days. Happened when it was on because  it was for most of the time. I will try to increase the polling period to 10 now. See what happens

teymur1988, 2020-02-12:
Update:
@RavenSystem increasing it to 10 didn't help either. After some random time it went into sensor error mode. Any suggestions? I've put it back to 5. If there is no direct solution to this, can it be coded to reboot on sensor error? If yes then how? How do you reboot a device running HAA? Thanks.

Regards

Teymur

teymur1988, 2020-02-16:
@RavenSystem Any suggestions?

RavenSystem, 2020-02-19:
Actions `"5"` is exec when sensor fails. You can reboot your device adding this action:

teymur1988, 2020-02-19:
@RavenSystem Thanks for the hint! Do you have any clue why is this happening to me with my setup?
You know I've changed the config of the thermostat to be Temperature and Humidity sensor with the same polling interval and it's rock solid! No failures. For some reason themostat fails after some random time :(

RavenSystem, 2020-02-19:
I don't know reasons for that failure. Maybe some kind of interference when relay is triggered. Software part used to read temperature and humidity sensor is the same for Thermostat and Temperature and Humidity sensor. I use a Sonoff TH16 without any issue.

teymur1988, 2020-02-19:
Thanks for your help! I also was wondering the same! Maybe TH16 has got some kind of isolation between the sensor data lines and the relay power lines... But it's strange how Temp&Humid sensor works all the time and thermostat doesn't. For now I'll try your suggestion with rebooting when sensor fails, lets see what happens. Thanks again.

teymur1988, 2020-02-27:
Hello again @teymur1988 
That rebooting command on sensor error made my thermostat a bit weired. Here's what happened: I guess once it got rebooted I was seeing the thermostat in Homekit without any values, it wasn't showing the temperature at all. So I decided to change the sensor, I bought a brand new DHT22. (I decided to change the sensor because I did change the sonoff device before). So what I'm observing now is that it kinda works as it should I see no sensor errors but the thing is that what I'm pretty certain about is that when temperature stays the same for a long period of time and eventually reaches the target heating temperature the thermostat doesn't switch the output off but it shows in homekit that it's cooling down. This has been happening for sometime now that's why I'm writing. Another evidence is that I've got a second sonoff with the old DHT22 undergoing a test. That is working properly because what I've got is an small LED lamp and the sensor underneath it so that the temperature is constantly changing and thermostat is properly working. So what do you suggest to do if in the real case the room temperature doesn't fluctuate that often? Here are few pics on how it looks over here: https://ibb.co/g6LTfg3
https://ibb.co/2tZv03R

teymur1988, 2020-03-07:
Hi @RavenSystem ,
I think I figured out my issue with this. The problem was related to the cooling state not having any output at all. So whenever target temperature was reached the "cooling" started in software but the hardware part for heating was staying on especially when the target temperature was reached after a long time. So now what I've done I've included Cooling state output to switch off the relay. Here is the config. Nad yes I added temp & humid sensor just for fun as well.

`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":0}]},"3":{"r":[{"g":12,"v":1}]},"t":21,"b":[],"g":14,"n":2,"j":5,"w":3,"m":20,"x":30,"s":5},{"t":24,"b":[],"n":2,"g":14,"j":5}]}`

closed 2020-03-09 by teymur1988

i3laze, 2020-06-26:
Weird thing.  I've just received AM2301 sensor from Sonoff Official Store at AliExpress..
It has both AM2301 Sonoff written on back side and inside on green breadboard (seen through the holes).

And it works only with setting Sensor Type = Si7021, not DHT22.
Anyone to confirm same behavior?