RavenSystem/esp-homekit-devices #786: Very strange problem

peros550, 2020-02-07:
At home I have 4-5 HAA devices, a couple Wemos D1 mini running custom HomeKit firmware, a few accessories based on Homebridge plugins and some Philips Hue. 

Today , for no apparent reason, all my HAA devices were disappeared from Home app. The wemos D1 devices running custom HomeKit firmware are listed but in red no response status. 

The only devices that are still operational in home app are Philips hue and all Homebridge based plugins. Have already tried rebooting router and Apple TV with no luck. Any ideas?

All HAA devices are connected to the WiFi ok.

RavenSystem, 2020-02-07:
Disappeared from Home app? Removed?

peros550, 2020-02-08:
Yes, HAA devices were completely removed. Other custom homekit devices were not removed, but they were unreachable (red no response). 

But hang on there is more. Situation is more complicated than I originally thought. The same issue happened to my second house. For reference, note that the two houses are connected the past 6 months with a site-to-site VPN using two Asus RT-AC68U. 

At either Apple TV homekit page, you could see both ATVs listed there as hub. Of course, I had disabled the remote ATV as hub for each location to avoid collisions. But sometimes this setting kept changing by itself ending up both Apple TVs listed as homekit hub. 

Thanks God, you have implemented the reset Homekit ID function, in this way I was able to bring back a device without re-flashing.

Here is a log taken during the issue: 
```
MacBook-Pro:~ yiannispapoutsis$ dns-sd -B _hap._tcp local
Browsing for _hap._tcp.local
DATE: ---Sat 08 Feb 2020---
10:32:20.965  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
10:32:20.966  Add        3   8 local.               _hap._tcp.           Philips hue - 289693
10:32:20.966  Add        3   8 local.               _hap._tcp.           HAA-DA5DB8
10:32:20.966  Add        3   8 local.               _hap._tcp.           HAA-A39537
10:32:20.966  Add        3   8 local.               _hap._tcp.           HAA-7FF787
10:32:20.966  Add        3   8 local.               _hap._tcp.           HomekitSensor-14E444
10:32:20.966  Add        3   8 local.               _hap._tcp.           HAA-2BE052
10:32:20.966  Add        2   8 local.               _hap._tcp.           HomeBridge-3188  
```

Most probably the issue caused by me due to the special connectivity I had implemented, but on the other hand the question remains, is it really possible for homekit framework to remove a device  from home app without user interaction? Any ideas, or suggestions more than welcome :)

PS: Sorry no logs from HAA devices, but if really needed I can open a sonoff S20 to connect a serial.

RavenSystem, 2020-02-08:
There is nothing in HAP that can remove accessories without user interactions.

I think it is an Apple HomeKit issue. When you remove a HAA accessory from Home App, HAA automatically reset HomeKit ID, then you can pair it again without enter setup mode to do that. But in this case, HAA devices keep pair info.

I suggest you to remove your HomeKit houses and start from scratch.

peros550, 2020-02-10:
Further testing and observation revealed that somehow Homekit kept restoring the status of home accessories to a previous state in which those HAA devices were not even added yet. 

That's why my initial observation was that HAA devices were removed. I encountered this problem more than once during the weekend. 

Definitely not a HAA firmware related issue, but I just wanted to share my experience. 

I will try some less intrusive actions (like deleting unused scenes, automation rules, etc) but if I fail to  resolve the issue, I will have to delete Home as you suggested.

closed 2020-02-10 by peros550