RavenSystem/esp-homekit-devices #787: Connection quality problem

quartist, 2020-02-07:
Hi, 

my sonoff T3 EU 3CH with HAA 1.5.8 time to time lost connection and stuck in loop like:

connect-dhcp assign-disconnect.
Logs from router:
dhcp3 deassigned 192.168.130.130 from 85:0D:88:77:77:D0
dhcp3 assigned 192.168.130.130 to 85:0D:88:77:77:D0
85:0D:88:77:77:D0@BSw10: disconnected, received disassoc: sending station leaving (8)

when I force setup mode (8times click) or plug out/in from power after that everything works correctly. Sonoff connects to wifi and work in Home app.

--

macjl, 2020-02-07:
I also have random loss of ping of my haa devices. I didn’t opened an issue yet, because I didn’t capture logs.
For me, I can see this behavior when I loose ping to a device more than 10 minutes. It’s randomly appear on any of my 14 haa devices about once a day. 

I don’t know if it’s the same problem, but may be related.

quartist, 2020-02-07:
Wireless client stats for this sonoff T3 EU 3CH : https://ibb.co/LdXygzQ
if 1 - works ok.
if 0 -device is unreachable. (looping, ping no working)

marcoscv-work, 2020-02-10:
As far as I understand HAA does not manage the Wi-Fi connection.
Some device issues maybe?

quartist, 2020-02-10:
Hi,

what do you mean "not manage the Wi-Fi connection"?
HAA connecting to wifi, selecting channel, assign IP address...

yusufd1, 2020-02-14:
Try 1.6.4 f/w many of the wifi issues are now resolved. It is much improvement from old f/w

quartist, 2020-02-15:
Hi,

soooo many changes! Update in progress...

closed 2020-03-16 by stale[bot]