RavenSystem/esp-homekit-devices #788: Sonoff mini Smart Doorbell

mientki, 2020-02-08:
I would like to use the sonoff mini and wall switch doorbell to make a smart doorbell. I would like to have 2 accessories (tiles) in the homekit: momentary switch for 1 sec and Occupancy Sensor. Activating for 10 seconds after pressing the button on the wall.

Occupancy Sensor because because it allows you to set a notification on your phone when it is triggered. I would like to know when someone rings my door when I'm not at home.

What is wrong with my config?

 `{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1,"i":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":6,"b":[{"g":0}]}]}`

Lubanski, 2020-02-17:
Do you resolved this issue?

RavenSystem, 2020-02-19:
It is a little mess. Forget HomeKit switch and better use this:

`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":6,"i":10,"1":{"r":[{"g":12,"v":1,"i":0.5}]},"b":[{"g":4,"t":0}]}]}`

closed 2020-02-26 by RavenSystem

allenmehrer, 2020-04-04:
I'm looking to do the exact same thing with the Sonoff mini. What were you able to achieve? Where you able to identify the mini as a doorbell in HomeKit?