RavenSystem/esp-homekit-devices #789: Feature Request: ability to use ESPOTA to update boot loader and HAABOOT OTA

tonysprenk, 2020-02-09:
Would it be possible to include something like espota in the firmware to have the ability to update the the boot loader and HAABOOT without soldering? Or would it take up too much space? 

https://github.com/esp8266/Arduino/blob/master/tools/espota.py

RavenSystem, 2020-02-09:
When you update HAA, HAA OTA (You call it HAABOOT) is updated too. And rboot (boot loader) is not necessary to update because there is not any new version used by esp-open-rtos. When a new version comes, I will update HAA OTA to include that feature.

There are 2 firmwares installed into your device: HAA and HAA OTA.

tonysprenk, 2020-02-09:
Thanks for the explanation. That makes a lot of sense!

closed 2020-02-09 by tonysprenk