RavenSystem/esp-homekit-devices #790: Strange behavior on GPIO inputs

ArFurik, 2020-02-09:
I try to create a motion sensor with a NodeMCU ESP8266 v3. Your software is great and the JSON able to set all I want... but a strange behaviour occurs on the GPIOs used as inputs. I tried several GPIOs and the problem is the same :
- When the input switch to hight, the signal is detected, I can see a led lighting ans the Home app say the a motion is detected. The itching at 7 returns all theses states to 0 -> **OK**
- If an other 1 is applied during the 2 seconds (approx.) after, all of this good behaviour repeats -> **OK**
- If an other 1 is applied more than 2 second after -> **NOTHING**, **_but_** the following 1 at the input is well detected, no matter the time elapsed then.

The detection seems done once in two.

I tried several GPIOs, I tested with a physical switch or my motion detector, putting or not pulldown resistors. This strange behaviour is exactly the same in all the cases.
I spent a lot of time trying to adapt the JSON, without solving the problem.
I sought on the net if this could be an electronical behaviour from the ESP or the NodeMCU, but nothing about that.
So I think it is a software issue.

Thank you in advance for your help, the JSON used is paste below.

`{
  "c": {
    "l": 13,
    "b": [{"g": 4,"t": 3}]
  },
  "a": [
    {
      "0": { "r": [{"g": 12,"v": 0}] },
      "1": { "r": [{"g": 12,"v": 1}] },
      "t": 12,
      "i": 7,
      "f1": [{"g": 14}]
    }
  ]
}`

vsidlu, 2020-02-09:
try this JSON - {"c":{"o":1,"l":2,"i":1,"b":[{"g":0,"p":1,"t":1}]},"a":[{"t":12,"i":5,"f1":[{"g":12,"p":0,"i":1,"t":0}]}]}
which PIR model are you using? do not forget that esp8266 need PIR to use 3v.

ArFurik, 2020-02-10:
Hi, it's a PIR HC-SR501. No problem with it and its output is a TTL signal 3.3v. I tried both with a switch or the PIR and it is exactly the same behaviour.
I will adapt your JSON to my GPIO config and try.

ArFurik, 2020-02-10:
**It's work !!!** _Thank you Uldis_
Some little things are not understood in the JSON commands yet for me.
I put this JSON : 
`{"c":{"l":13,"b":[{"g":4,"t":3}]},"a":[{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":12,"i":7,"f1":[{"g":14,"p":0,"i":1,"t":0}]}]}
`
The difference is on the f1 section : 
` "f1": [ {
          "g": 14,
          "p": 0,
          "i": 1,
          "t": 0
        } ]`

- "p":0 -> So it seems that a pullup resistor is the default and I must indeed disable it.
- "i":1 _and_ "t":0 -> This is more mysterious... Why may I have to invert the input signal... Well in this case, with i and t, it could be double-inverted and so it works.

Whatever, all works now perfectly, and I **thank you very much**. I spent a lot of time on this before posting a issue.

closed 2020-02-10 by ArFurik

vsidlu, 2020-02-10:
happy to help!!!

"i":1 is triggering motion sensor when input is low. if you use "i":0, then triggering will be when input is back to high which is when motion sensor gets back to no motion sensed. so with "i":1 motion message appears almost instantly.

"t":0 also for me is not 100% clear, but @RavenSystem pointed it out in one of opened issues and it helped. i think it is replacing boolean used in older HAA versions.

ArFurik, 2020-02-10:
Ok, I'll keep that in mind for my future HAA projects.
Have a nice day.