RavenSystem/esp-homekit-devices #794: Setup mode: Force WiFi BSSID

i3laze, 2020-02-11:
Roaming decisions in WiFi stack are made solely by clients. Network can only assist by using technologies like 802.11k,r,v or just kick a client from an AP.

Any support of these in current ESP implementation?
Is there generic polling and roaming based on a signal strength?

What will happen if a nearest AP goes down when Force BSSID is enabled?
Without waiting an answer, I suggest adding a third mode:
- Force BSSID, while AP is online.
- When AP goes offline, connect to nearest AP with same SSID and try reconnecting to main AP every 5/10 min.

yusufd1, 2020-02-14:
In the latest 6.1.4 now device searches for best wifi signal (not necessarily only based on rssi) and connects to that signal so no real use case for force bssid anymore unless user wants his device to connect to a specific AP.

closed 2020-03-15 by stale[bot]

i3laze, 2020-05-12:
Right now I'm troubleshooting WiFi with Sonoff Basic and latest HAA 2.3.1..

Force BSSID mode enabled but not reachable (I've put a fake BSSID).
How long till device goes Emergency Setup Mode with HAA-SSID broadcast?