RavenSystem/esp-homekit-devices #796: No response

asbesto01, 2020-02-11:
Ciao to everybody,

I have always had a rock solid ravencore setup (1 year more or less) with:

6 blinds (sonoff 4ch pro r2 on Ravencore 1.1.1)
1 light switch on the balcony (sonoff basic on Ravencore 1.0)
1 lock (sonoff 4ch pro r2 on ravencore 1.1.1)

1 temp and humidity sensor (sonoff th16 on Ravencore 1.0)
2 lights in the bedroom (sonoff basic on Ravencore 1.1)
1 switch in the kitchen (sonoff basic on Ravencore 1.0)
light control (2 sonoff 4ch pro r2 on Ravencore 1.2)

They were extremely reliable, zero issues at all!!

Since some days, blinds, lock and balcony light are becoming crazy NO RESPONSE.
The other sonoff devices are ok, without any issues.

I wake up in the morning, some of them are in NO RESPONSE. I reboot router, sometimes it fixes, sometimes it fixes some blinds but other ones are in NO RESPONSE.

I tried:

1) power cycling all the house, it solves but only for some hours, then back again.
2) changing HomeKit hub, from my iPad to my Apple TV. Nothing changes.
3) I thought that zigbee devices could interfere with wifi quality, so I forced the router to work only on band 1, while zigbee devices work on bands from 1 to 13. No solutions
4) I forced the router to work on band 13 with Russia (and then Japan) specifications (I am in Europe) so I could avoid all wifi interferences with my neighborhoods. No luck

What I am thinking is:

1) re-flash all the ravencore to HAA. It's an enormous pain in the ass because all the sonoff are hidden inside walls so I have to destroy half of the house.
2) move the router away from the cabinet where it is located now, far from the zigbee repeaters (I have 3; Philps Hue, Ikea Tradfri, Aqara Hub)

Any other suggestion would be extremely appreciated; I am tired and my wife is complaining MASSIVELY.

Thank you!

yusufd1, 2020-02-14:
Have you changed anything in your network recently maybe added any access points or range extenders especially on the 2.4ghz range?

yusufd1, 2020-02-14:
Also when they go to no response and if you look on your client list in your router webpage are the devices that are on no response still connected to wifi?

asbesto01, 2020-02-16:
Ciao Yusuf, 

the only major changes have been:

1) introduction of different iPad. Swapped to the old one, still no solution
2) introduction of two zigbee hubs. shut them down, still no solution

To answer you: when it's no response, looking at router control panel the blind is NOT connected to wifi. After some minutes (or hours) they connect back again!

Any ideas?

gnoto, 2020-02-17:
I have the same problem.
I just finished flashing all the ravencores on HAA but apparently with no advantage.
Homekit instability?

yusufd1, 2020-02-18:
Hi not sure what could be the issue if nothing was added to the network. My problems started with APs being added and if devices are too far from router. Usually the no response issue is if the WiFi signal is not great or if the device is connected to WiFi then the mdns announcement was not perfected which HAA 1.6.4 has kind of resolved.

asbesto01, 2020-03-16:
> Hi not sure what could be the issue if nothing was added to the network. My problems started with APs being added and if devices are too far from router. Usually the no response issue is if the WiFi [...]

After 30 days with more or less zero issues, today, AGAIN!!

Still on ravencore 1.1.1

Jose RavenSystem I need your help.. these 5 blinds are the latest ravencore devices not migrated to HAA, because no one is able to help me with the config 

(Please see post for reference):

I am sure you can help me, and I am sure that the answer is in ravencore vs HAA because ALL my other HAA devices (50+) are perfect!

Help!

asbesto01, 2020-03-30:
Update:

Replaced two of the sonoff 4ch pro r2 blinds, which were running on ravencore, with two brand new sonoff dual running HAA.

the problem is still there.

I am going nuclear by replacing the router with a mesh one —> fritzbox 7590 with fritzrepeater 3000

After that I don’t have anymore solution.

I have tried everything included iPhones replacement 

So sad

Retireddad, 2020-03-30:
Check your router setting for 2.4g, i.e. is it set for 802.11 b/g or 802.11 b/g/n?

asbesto01, 2020-03-30:
It’s set 802.11 n only

How should I set it?

I highly doubt this is the cause, because I have dozens of identical devices and only two of them show the same issue

Retireddad, 2020-03-30:
802.11 b/g are for 2.4g
802.11 n is for 5.8g, so if your 2.4g wifi has the n setting it will also try and connect to 5.8g which seems to confuse HAA devices. Try using only 802.11 b/g setting in 2.4g

Retireddad, 2020-03-30:
11a         5 GHz
11b         2.4 GHz
11g         2.4 GHz
11n         2.4 GHz and 5 GHz
11ac        5 GHz

asbesto01, 2020-03-30:
I have following settings

802.11n only ( what I am using now)
802.11g/n mixed
802.11 b/g/n mixed

Which one should I select?

However, At the moment using the first one I can ping dozens of identical 2.4 ghz devices except two of them. They go in no response sometimes and I cannot understand why

Retireddad, 2020-03-30:
I'd definitely give the other 2 settings a try, shame they all have the "n" parameter. Mine allows me to select either 802.11 b/g/n or 802.11 b/g for 2.4g.

asbesto01, 2020-03-30:
Tried for 6 hours, not solved :(

Retireddad, 2020-03-30:
What router are you using?

asbesto01, 2020-03-30:
TP-Link touch p5

Ordered a fritzbox 7590 with mesh fritzbox repeater 3000; I will swap it hoping it solves :(

macjl, 2020-03-31:
I had similar issues whith multiples Wifi AP with same SSID. The HAA devices should use the better AP with the SSID configured, but with my test, it does'nt work well. I lose pings, and many HAA devices were in 'no response' in Home app. Other devices than HAA had not these problems, on same Wifi network.

For me, I solved my issue by having unique SSID on each AP and distribute devices on these AP. But that's a workaround, because if an Wifi AP fails, the devices will not connect to the running one, even if the signal strengh is enough.

So you could try to use different SSID on each AP you have to see if your issue is the same.

closed 2020-04-30 by stale[bot]