RavenSystem/esp-homekit-devices #798: JSON for WATER VALVE

pilot1981, 2020-02-12:
Hi,
I'm looking to create my new water valve using HAA... my project requirements are these:
- to control n.2 physical water valve 24VAC I would to use n.2 shelly 1 -> Shelly 1, relay control on GPIO4
- dummy switch to enter in setup mode easily
- config duration to open water valve in EVE app: 5, 10 30 minutes
- config schedule in EVE app: I would schedule start at 23.00 only on monday, wednesday and friday; but I don't found how to do it using timer of EVE app...
- OPTIONAL: I would add a rain sensor...but I don't know insert rain sensor in serial with physical water valve or add rain sensor to home kit (water sensor it's ok for me); in the last case I could attach ti to one or both GPIO0 of shelly 1 and create an automation in apple home app?

please see example in this schema diagram:
[image]

At moment I started using this JSON:

{"c":{"b":[{"g”:4,"t":5}]},"a":[{"t":20,"f0":[{"g”:4,"t":1}],"f1":[{"g”:4,"t":2}]}]}

Please can you:
- suggest best JSON;
- suggest best solution to schedule water valve starting in different days?
- suggest best solution to use rain sensor?

pilot1981, 2020-02-12:
This JSON could be good:

{"a":[{"t": 20,"d": 3600,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]}]}

But it don’t have dummy switch to enter setup mode, I don’t know best way to automate start of relay, I’m not sure howto use rain sensor...

RavenSystem, 2020-02-19:
> suggest best JSON?

`{"a":[{"t":20,"d":3600,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}`

> suggest best solution to schedule water valve starting in different days?

Use Apple HomeKit automations.

> suggest best solution to use rain sensor?

Use a Sonoff with a rain/water-leak sensor, and use its status as condition with the HomeKit Automations.

pilot1981, 2020-02-19:
Thanks, but why I can see water valve in Apple automation? Any idea?

RavenSystem, 2020-02-19:
I understand that you are talking about Home App. Apple does not support all automation kinds with its Home App. You must use a third party HomeKit App, like EVE or Home+.

PS: I edited JSON to change valve type.

pilot1981, 2020-02-27:
Hi, 
thanks for reply.

I confirm this JSON work fine:

{"a":[{"t":20,"w":1,"d":3600,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}

I'm testing EVE timer config to schedule water valve start...

@RavenSystem can you add to this JSON a stateless button for external/physical button to start water valve manually?

pilot1981, 2020-03-06:
@RavenSystem any help?

pilot1981, 2020-03-10:
i don’t understand why GPIO5 is already used in JSON config...

pilot1981, 2020-03-10:
I can add this string?

 "a": [{
    "t": 0,
    "f0": [{ "g": 5, "t": 1 }]
  }]

And stateless button start water valve with default duration?

pilot1981, 2020-03-22:
Hi,
@glp71 suggested to use this JSON:

{"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"t":20,"w":1,"d":3600,"b":[],"s":0},{"1":{"s":[{"a":1}]},"i":0.5,"b":[],"s":0},{"t":3,"b":[],"s":0,"f0":[{"g":5,"t":0}],"f1":[{"g":5,"t":0}],"f2":[{"g":5,"t":0}]}]}

But stateless button isn’t able to select sprinkler....

[image]
[image]
[image]
[image]

It isn’t possible to use GPIO5 for physical external button?

pilot1981, 2020-03-22:
This one can work for me?

{"a":[{"t":20,"d":3600,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"i":0.5,"1":{"s":[{"a":1}]},"t": 0,"f0": [{ "g": 5, "t": 1 }]}]}

pilot1981, 2020-03-22:
Or this one could be better?

{"a":[{"t":20,"d":3600,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"i":0.5,"1":{"s":[{"a":1}]},"f0": [{ "g": 5, "t": 1 }]}]}

pilot1981, 2020-03-23:
Solved with this JSON:

{"a":[{"t":20,"d":3600,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}

closed 2020-03-23 by pilot1981

KrzysiekRadek, 2020-05-09:
Hi All :)
I need a sprinkler string for Sonoff BASIC ...
can you help me?