RavenSystem/esp-homekit-devices #799: I've tried configuring the Shelly1 with the JSON string that was proposed above, but I'm unable to press the button in the iOS Home app. What would be the correct string so that I can either use the momentary physical push button connected to the SW pin on the Shelly (GPIO5) or use the Home app to trigger that button? Action is then to momentarily short pins I and O on the Shelly1 (GPIO4). I'm basically trying to control a garage door by connecting the current push button to Homekit. Right now, 1 press on that button activates the door, another press stops it, another press activates it again but in the other direction, etc and all this happens in the garage actuator. I just need the Shelly to short O and I whenever the physical button is pushed or whenever I push the button in the Home app.

HAAnewbie, 2020-02-16:
I've tried configuring the Shelly1 with the JSON string that was proposed above, but I'm unable to press the button in the iOS Home app. What would be the correct string so that I can either use the momentary physical push button connected to the SW pin on the Shelly (GPIO5) or use the Home app to trigger that button? Action is then to momentarily short pins I and O on the Shelly1 (GPIO4). I'm basically trying to control a garage door by connecting the current push button to Homekit. Right now, 1 press on that button activates the door, another press stops it, another press activates it again but in the other direction, etc and all this happens in the garage actuator. I just need the Shelly to short O and I whenever the physical button is pushed or whenever I push the button in the Home app.

_Originally posted by @HAAnewbie in https://github.com/RavenSystem/esp-homekit-devices/issues/411#issuecomment-581131376_

peterduyts, 2020-02-16:
I use this on my shelly1 for the garage door:  

{"a":[
{
"t": 1,
"d": 1,
"0":{"r":[{"g":4}]},
"1":{"r":[{"g":4,"v":1,"i":1}]},
"b":[{"g":5,"p":0},{"g":5,"t":0}]}]}

It's a switch that returns to off afer 1 second ( "i":1 )  . I do not use the GPIO5 for the physical button, but just connected it in parallel over the shelley relay pins so that in case the shelly dies I can always keep using the physical button.  Of course it gives no fancy animation in homekit but it does the job perfectly and you can make a scene that just switches this on and call it "Open Sesame" ;)

HAAnewbie, 2020-02-16:
It worked like a charm, thanks a lot! And great idea to keep the physical switch in parallel over the Shelly relay pins!

closed 2020-02-16 by HAAnewbie

Skankyhank, 2020-02-25:
Any chance to get a picture or detailed description of the exact wireing goin’ on here? :)
I’m about to do the same thing, I have a LimusOne G80 garage door opener.

Skankyhank, 2020-02-26:
> It worked like a charm, thanks a lot! And great idea to keep the physical switch in parallel over the Shelly relay pins!

Are you able to provide some information?