RavenSystem/esp-homekit-devices #800: RGBWW Bulbs

rasod, 2020-02-16:
Is there a way to configure a RGB bulb that has 2 white channels?

I got some RGB bulbs and flashed (1.6.4) one using Tuya convert. I configured using what I think is the correct pins for the channels, however the bulb doesn’t really change colours and when you turn off the bulb it just changes colour temperature. I think this is because the configuration is only turning of one of the white channels off and the other is remaining on always

Configuration I have used:

```{"a":[{"t":30,"r":4,"g":12,"v":14,"w":5,"s":1}]}```

Below is the Home Assistant ESP Home Configuration for what I think is the correct bulb

```
esphome:
  name: garagebulb1
  platform: ESP8266
  board: esp01_1m

output:
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO12
    max_power: 0.7

  - platform: esp8266_pwm
    id: output_red
    pin: GPIO4
    max_power: 0.7

  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO14
    max_power: 0.7

  - platform: esp8266_pwm
    id: output_wwhite
    pin: GPIO13
    max_power: 0.65

  - platform: esp8266_pwm
    id: output_cwhite
    pin: GPIO5
    max_power: 0.65

light:
  - platform: rgbww
    name: "garagelight1"
    id: light_rgb
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cwhite
    warm_white: output_wwhite
    cold_white_color_temperature: 5000 K
    warm_white_color_temperature: 2000 K
    effects:
      - strobe:
      - flicker:
      - random:
# Ensure the light turns on by default if the physical switch is actuated.
    restore_mode: ALWAYS_ON
```

DanielSergienko, 2020-02-17:
As It is in my knowledge, HAA only supports RGBW lights, so the only opion i see here is to divide i.e. warm white either cold white into another virtual lightbulb or simple switch what will control it.
Something like that:
`{"a":[{"t":30,"r":4,"g":12,"v":14,"w":5,"s":1,"p":1024,"d":1,"e":10,"fr":1,"fg":1,"fv":1,"b":[]},{"t":30,"s":0,"p":1024,"d":1,"e":10,"fr":1,"fg":1,"fv":1,"b":[],"w":13,"fw":0.01}]}`

magichew, 2020-02-17:
Hi

I created this thread a while ago to ask about this... https://github.com/RavenSystem/esp-homekit-devices/issues/638

I've come from a homebridge set up where I did have support for RGBWW. The set up I had was Homebridge, Domoticz and e-Domoticz plugin for HomeBridge.

RGBWW bulbs worked as you would expect with colour and hue control.

I've moved to HAA because I'm trying to move away from hubs. I don't have any 'bulbs' as such but I do have an RGBWW LED controlled. My work around for now is to disconnect one of the wires from the LED controller otherwise the signal was for always on for the wire that didn't have a GPIO assigned to control it and so was impossible to turn off via app control, only via turning off at the mains.

As commented on https://github.com/RavenSystem/esp-homekit-devices/issues/638 @RavenSystem has said that he needs an algorithm to convert one way of referencing colours to another way but I haven't been able to help.

"I only need an algorithm that converts HSI values to R G B CW WW."

So if you can help out with that then I think that will sort it.

rasod, 2020-02-17:
Thanks, didn't spot the other issue. I'll do some research

RavenSystem, 2020-02-19:
Duplicated: https://github.com/RavenSystem/esp-homekit-devices/issues/638

closed 2020-02-19 by RavenSystem