RavenSystem/esp-homekit-devices #806: Shelly 1 HAA & lock mechanism

Ni88-dpb, 2020-02-19:
Hello to all. I have a shelly1 with ravencore because I use it as a lock for the gate. How can I configure it with haa? It's the only device I still keep with ravencore because I can't set it ... thanks to anyone who can help me!

RavenSystem, 2020-02-19:
Please, take screenshots of your RaveCore settings.

Ni88-dpb, 2020-02-19:
> Please, take screenshots of your RaveCore settings.

Here you are

[image]

[image]

Ni88-dpb, 2020-02-19:
Thanks, I managed to set haa as a button and it works, but I can't replicate the lock. on homekit it is important because I ask siri to "open the gate" and if I set the shelly as a standard button, siri does not open it.

RavenSystem, 2020-02-19:
If you look at https://github.com/RavenSystem/esp-homekit-devices/wiki/Devices-Database there is a Shelly 1 configured as "Lock Mechanism with external push".

closed 2020-03-20 by stale[bot]