RavenSystem/esp-homekit-devices #807: Sonoff Dual R2 as window covering (roller blinds)

teymur1988, 2020-02-19:
Hi @RavenSystem 
I've just got Sonoff Dual R2 for my roller blinds. I can't seem to be able to set it to come to "Last State" after reboot... I tried setting "s": 5 but that didn't help. I've read the wiki and couldn't find any info about it, and haajc v0.6 doesn't have window covering template. Would you be so kind to help?

teymur1988, 2020-03-09:
Hi @RavenSystem ,
This is yet another thing I came accross when I just installed HAA on a Sonoff Dual R2 and configured as window covering. This is my config:
`{"c":{"l":13,"b":[{"g":10,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1},{"g":5}]},"1":{"r":[{"g":12},{"g":5,"v":1}]},"2":{"r":[{"g":12},{"g":5}]},"3":{"r":[{"g":12,"i":1},{"g":5}]},"4":{"r":[{"g":12},{"g":5,"i":1}]},"t":45,"o":87,"c":87,"f":18,"f2":[{"g":10,"t":0}]}]}`

When I want to enter the setup mode by quickly toggling in homekit the sonoff turns both relays on!!! Luckily I didn't have the motor connected to it otherwise it would have burned it completely. Is there anything wrong in the config? When I physically hold the button on sonoff (GPIO10) for a long time it does go to setup mode. How can I enter setup mode by toggling quickly in homekit? O yeah and the previous message is also still intact... How do I make it to go to last state after a power outage? Because say if the covering is open 50% after power outage it will show in homekit as closed!

Regards
Teymur

teymur1988, 2020-03-09:
Just in case you're wondering about `"f2":[{"g":10,"t":0}]` this. I have 2 reed switches connected in parallel on fully open and fully close positions to stop the motor.

ericklil, 2020-04-01:
Is there a way to dynamically update the `"o" `, `"c"` & `"f"` function. This to facilitate set up of window size...  Or on a broader scale can one update any of the `JSON` variables after a unit has been initialized.

asbesto01, 2020-04-10:
> Hi @RavenSystem
>
> I've just got Sonoff Dual R2 for my roller blinds. I can't seem to be able to set it to come to "Last State" after reboot... I tried setting "s": 5 but that didn't help. I've read [...]

Exactly same issue. After power outage they would always restart in CLOSED mode.

This is my code:

{"c":{"l":13,"b":[{"g":10,"t":5}]},"a":[{"t":45,"o":22,"c":22,"f":0,"0":{"r":[{"g":12,"v":1},{"g":5}]},"1":{"r":[{"g":12},{"g":5,"v":1}]},"2":{"r":[{"g":12},{"g":5}]},"3":{"r":[{"g":12,"i":1},{"g":5}]},"4":{"r":[{"g":12},{"g":5,"i":1}]},"f3":[{"g":0}],"f4":[{"g":9}]}]}

teymur1988, 2020-04-11:
Hi @asbesto01 
Did you try to update the HAA version? The issue seems top be fixed in one of the recent firmware releases. I've not tested myself tho.

Teymur.

asbesto01, 2020-04-11:
On which version? I am on 1.8.4 at the moment!

RavenSystem, 2020-04-11:
Fixed on v1.9.0.

closed 2020-04-11 by RavenSystem