RavenSystem/esp-homekit-devices #809: Changed router - how to set up again ?

macguru341, 2020-02-21:
Hello
I’ve changed my router and have a new WiFi network with a new name .
I had the sonoff touch (3 lights) connected in the past to HomeKit and everything worked fine but now that I’ve changed the WiFi network it’s not working . 
What I need to do to connect it back to HomeKit again ? I hope I don’t need to remove the switch from the wall again 😕.

tonysprenk, 2020-02-21:
Easiest way is to change your new network SSID to be the same as your old SSID. Alternatively you can enter setup mode and if the device can’t connect to the specified SSID it will create its own AP which you can connect to. More in this can be found in the wiki: https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode

macguru341, 2020-02-21:
> Easiest way is to change your new network SSID to be the same as your old SSID. Alternatively you can enter setup mode and if the device can’t connect to the specified SSID it will create its own AP which you can connect to. More on this can be found in the wiki: https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode

Thanks - how can I put into setup mode?

tonysprenk, 2020-02-21:
Read the wiki.

macguru341, 2020-02-21:
> Read the wiki.

I’ve read it , and it says toggle on if the switcher buttons 8 times , but nothing happens , I’ve searched in my WiFi on my iPhone and I can’t see the switch APi

tonysprenk, 2020-02-21:
Then you must enter emergency setup mode.

DanielSergienko, 2020-02-21:
@macguru341 emergency setup mode:
plug it in, wait 2 seconds. Unplug. Plug again. 
If you had configured led on it, first time you plug it it should blink twice. When you replug it - only once. 
Keep trying until you see HAA-XXXXXX network.

macguru341, 2020-02-22:
Now I remembered that I’m using RavenCore v1 so the go into setup mode is the same?

macguru341, 2020-02-22:
Ok got it :
1. Turn off the power for at least 5 Min
2. Turn on the power 
3. Press one of the buttons 10 times
4. Connect to RavenCore WiFi and enter the new WiFi network 
5. Wait a bit and the had the new accessories called Raven and than use the Eve app to configure it.

Thanks

closed 2020-03-10 by RavenSystem