RavenSystem/esp-homekit-devices #811: How to Config Wemos D1 mini just for Temperature and Humidity Sensor

sblitzer, 2020-02-21:
Hello,

Can someone please help me with providing JSON to  Temperature and Humidity Sensor only (No sonoff) (GPIO5) DHT22  Type 2, HAA flashed to Wemos D1 mini.

Thanks for your help

DanielSergienko, 2020-02-21:
```
{
  "c": {
    "l": 2
  },
  "a": [
    {
      "t": 24,
      "b": [],
      "n": 2,
      "g": 5
    }
  ]
}
```

mafyata, 2020-02-22:
I use this:
{"c":{"n":"Hol-TempHum","b":[{"g":5,"t":0}]},"a":[{"t":24,"g":14,"z":-1},{"0":{"r":[]},"1":{"r":[],"s":[{"a":1}]},"t":1,"b":[],"s":0}]}

sblitzer, 2020-02-22:
> ```
> {
>   "c": {
>     "l": 2
>   },
>   "a": [
>     {
>       "t": 24,
>       "b": [],
>       "n": 2,
>       "g": 5
>     }
>   ]
> }
> ```

Doesn't work :(

sblitzer, 2020-02-22:
> I use this:
> {"c":{"n":"Hol-TempHum","b":[{"g":5,"t":0}]},"a":[{"t":24,"g":14,"z":-1},{"0":{"r":[]},"1":{"r":[],"s":[{"a":1}]},"t":1,"b":[],"s":0}]}

Works perfectly thank you 👍

closed 2020-02-22 by sblitzer