RavenSystem/esp-homekit-devices #812: Tuya Smart Glass Touch Wall Switch

mriksman, 2020-02-21:
Hi,

These glass touch wall switches are very common (https://www.aliexpress.com/item/4000615022697.html) - and they all use the TYWE3S (ESP 8266 derivative) chip. I have successfully flashed the device using `tuya-convert`, and am now trying to use JSON to configure it. Surprisingly, I am finding this very difficult. Has anyone else configure a device like this that has Relay, LED (Red for relay ON, Blue for Relay OFF), and Button? This particular one also has a green status LED.

```
AVATTO US/AU Wall Switch
SYS-WS 01/02/03
GPIO13			Relay1
GPIO15			Relay2
GPIO4			Relay3
GPIO16			Button1
GPIO5			Button2
GPIO3 (RXD0)    	Button3
GPIO12			LED1
GPIO1 (TXD0)    	LED2
GPIO14 			LED3
GPIO0			Status LED
```

Questions
1. When ever a configured button is pressed, the Status LED briefly flashes. Can this be disabled?
2. How can I configure the Button feedback LEDs to properly show the state of the Relays? When I start the device, the LEDs are OFF (High-Z state). They should be Blue and Relays OFF on startup.

DanielSergienko, 2020-02-21:
To disable status led delete "l": LedGPIO in {c array

To help you more it would be nice if you could provide us your actual JSON config

mriksman, 2020-02-22:
Thanks.
I still like the Status LED for when you hold down the button for 8 seconds, and it flashes 3 times fast to let you know it is entering Setup Mode. It would also be handy to have this Setup LED flash if it can't connect to Wi-Fi...? It doesn't make sense to have it flash when pressing a button - surely that's something you would configure on a per-Button basis depending on user preference?

```
{
	"c": {
		"l": 0,
		"b": [{"g": 16,	"t": 5}]
	},
	"a": [
		{
			"t": 1,
			"0": {"r": [{"g": 13, "v": 0}, {"g": 12, "v": 0}]},
			"1": {"r": [{"g": 13, "v": 1}, {"g": 12, "v": 1}]},
			"b": [{"g": 16,	"t": 1}]
		},
		{
			"t": 1,
			"0": {"r": [{"g": 4, "v": 0}, {"g": 14, "v": 0}]},
			"1": {"r": [{"g": 4, "v": 1}, {"g": 14, "v": 1}]},
			"b": [{"g": 3,	"t": 1}]
		}
	]
}
```
As you can see - I am trying to have a button cause both the Relay (GPIO 13 and GPIO 4) AND associated LED (GPIO 12 and GPIO 14) to change state.

I need this LED to always represent the state of the Relay. How is this possible?

hillaliy, 2020-02-22:
I have Tuya Covering Touch Wall Switch.
This is my Json:
{"c":{"l":0,"i":0,"b":[{"g":5,"p":0,"t":5}]},"a":[{"t":45,"o":14,"c":13,"0":{"r":[{"g":13,"v":1},{"g":15}]},"1":{"r":[{"g":13},{"g":15,"v":1}]},"2":{"r":[{"g":13},{"g":15}]},"3":{"r":[{"g":13,"i":1},{"g":15}]},"4":{"r":[{"g":13},{"g":15,"i":1}]},"f0":[{"g":12,"p":0}],"f1":[{"g":5,"p":0}],"f2":[{"g":3,"t":0},{"g":3,"t":0}]}]}

mriksman, 2020-02-22:
Oh. Turns out my config works fine...! Nice. 

So the only thing annoying me now is the Status LED flashing when a button is pressed. I still want it for Wi-Fi feedback and all that, just not for when a button is pressed.

closed 2020-02-22 by mriksman