RavenSystem/esp-homekit-devices #814: Sonoff RF

DanielMosco, 2020-02-23:
Hello all. Please help me with JSON string. I have Sonoff RF and i want to use RF button. 
Thanks in advance.

DanielMosco, 2020-02-24:
Issue fixed.

closed 2020-02-24 by DanielMosco

pilot1981, 2020-02-24:
@DanielMosco can you share JSON?

DanielMosco, 2020-02-24:
@pilot1981 It's basically the first JSON in the device database. I just couldn't put the RF module into pairing mode because it was the older type (double click to start learning) instead of long press.  In case u need: 
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

pilot1981, 2020-02-24:
I’m interested: Can you explain better the pairing function?

I will receive soon a sonoff RF R3 and I have to pair it with DW1: What I have to do?

1) flash with HAA
2) config your JSON
3) howto pair with DW1?

DanielMosco, 2020-02-24:
Sorry but what is DW1?

pilot1981, 2020-02-24:
It’s RF sensor:

https://sonoff.tech/product/accessories/dw1

I would use it for my mailbox...I have problem using zigbee and WIFI sensors...

DanielMosco, 2020-02-24:
@pilot1981 I'm not  sure you can pair these to modules together. I paired RF button to the Sonoff RF and use it as external toggle.  Open the new issue as this one already closed by me and maybe not monitored by other members.

pilot1981, 2020-02-24:
I already tried with specific issue but no any answer...you are the more experienced on it!!!!

If I understood button control pairing feature of RF...

I can also try this procedure:
1) pair RF Sensor using original ITEAD firmware
2) flash with HAA (it don’t touch RF Module)
3) config JSON
4) HAA accessory receive signal from RF Module (from DW1 sensor)

This can work in your opinion?

DanielMosco, 2020-02-24:
Theoretically You're good to go, but practically, you will need to try your procedure in order to see if it works. As I mentioned previously, I'm not sure DW1 can be paired directly with Sonoff RF. From my understanding, these sonoff sensors work through RF bridge.