RavenSystem/esp-homekit-devices #818: Shelly 1 Garage door opener with 2 magnetic reed contacts

morphix9, 2020-02-26:
Hey guys,

has anyone tried to connect 2 magnetic reed contacts with the Shelly 1? One contact for fully open, one for fully closed.
Could someone share the JSON?
Which reed contacts do you prefer (where do you buy)?

fIctioN600, 2020-02-28:
Yes i do. Works without problems.

{"a":[{"t":40,"d":18,"b":[{"g":5}],"f3":[{"g":1,"t":0}],"f4":[{"g":1}],"f2":[{"g":3,"t":0}],"f5":[{"g":3}],"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2}}]}

Shelly must be connected to 24V DC Adapter.

I use this contacts:

[https://www.amazon.de/gp/product/B000ZGI1Y6/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1]

but any other NC contact will work to

morphix9, 2020-02-28:
Thanks @fIctioN600!
Could you tell me how do you connect the reed switches to Shelly (wiring)?

fIctioN600, 2020-03-02:
first reed: one wire to GPIO 1 and the other to GND
second reed: one wire to GPIO3 an the other also to GND

I used jumper cables to connect it.

morphix9, 2020-03-02:
> first reed: one wire to GPIO 1 and the other to GND
> second reed: one wire to GPIO3 an the other also to GND
> 
> I used jumper cables to connect it.

Thank you for your help, I will try it out.

kn1gh1, 2020-03-03:
What does it means? GPIO 1 and Gnd?

On my shelly only.    I O SW N L

fIctioN600, 2020-03-04:
Gpio 1 and GND are some of the connectors you use for flashing 
[https://images.app.goo.gl/U71bchp8VrsrtFD56](url)

kn1gh1, 2020-03-04:
Ah ok.  but why does it switch twice?  the gate opens and stops immediately with the second impulse.  can you take it out?  Are you German because the Amazon link is German?

fIctioN600, 2020-03-04:
I am german yes. It does not switch twice in my installation. First push open or close second push reverse direction there is no stop

kn1gh1, 2020-03-04:
I also come from Germany.  I tried again, my gate opens a little and then closes again.  Too bad I thought that's exactly what I'm looking for.  Did you write the json-config yourself?

kn1gh1, 2020-03-05:
funny, today my reeds get connected and working.  also switches only once, without reeds twice.  so everything is fine.

morphix9, 2020-03-30:
Working fine on my end. This JSON could be added to the library.

closed 2020-03-30 by morphix9

Cunner, 2020-04-15:
@fIctioN600 is it also possible to use NO reed contacts ? If so which parameters do I have to change?
Thanks in advance!

morphix9, 2020-04-20:
@fIctioN600 you mentioned the shelly needs to be connected to 24v. Why? Is the gpio output different when connected to 220v? I’m having an issue when on 24v, because the sensor wires are too long (20m).

fIctioN600, 2020-04-20:
Yes input = output and at least my contacts cant handle 220v for the relay I O at the shelly it doesnt matter at all

Gumboot13, 2020-07-16:
> output

Don't use the gpio pins while using 220v AC as the gpio pins are tied to the AC circuit and and could cause damage to you or your devices/sensors

Gumboot13, 2020-07-16:
> first reed: one wire to GPIO 1 and the other to GND
> second reed: one wire to GPIO3 an the other also to GND
> 
> I used jumper cables to connect it.

Only issue using gpio1 is if held low on boot/reset the device won't boot.

Gpio3 is fine,. Gpio4 can also be used (the SW input terminal)

Both can be held low on boot without issues.

morphix9, 2020-07-21:
@Gumboot13 thanks for that. I will switch then from GPIO1 do GPIO3.

Are you sure that the pins are outputting 220V when on 220V power supply? I measured the output with a multimtter, but got around 4V.

What about if you power with 24V? Is the pin voltage dependant on that?

Gumboot13, 2020-07-21:
> @Gumboot13 thanks for that. I will switch then from GPIO1 do GPIO3.
> 
> Are you sure that the pins are outputting 220V when on 220V power supply? I measured the output with a multimtter, but got around 4V.
> 
> What about if you power with 24V? Is the pin voltage dependant on that?

Have a read here http://www.thesmarthomehookup.com/the-shelly-1-smart-relay-is-it-better-than-the-sonoff-basic/

Apparently the gpio are tied to the AC side of it. On a plus side of using the 24v option then it is ok

morphix9, 2020-08-09:
> Only issue using gpio1 is if held low on boot/reset the device won't boot.
>
> Gpio3 is fine,. Gpio4 can also be used (the SW input terminal)
>
> Both can be held low on boot without issues.

How about using GPIO0 (instead of 4)? Can it be held low on boot?

Gumboot13, 2020-08-09:
> How about using GPIO0 (instead of 4)? Can it be held low on boot?

Can't use gpio0 as if you hold it low on boot places the device into flash mode

morphix9, 2020-08-09:
Right. Where the hell is the GPIO4 then on a Shelly 1? :)

Gumboot13, 2020-08-09:
> Right. Where the hell is the GPIO4 then on a Shelly 1? :)

It's the RX pin on the pin header located on the top back of the device

Gumboot13, 2020-08-09:
[image]

morphix9, 2020-08-09:
Could you mark on that image where GPIO4 is?
[image]

Gumboot13, 2020-08-09:
> Could you mark on that image where GPIO4 is?
> [image]

Sorry misread your post about gpio4, thought you were talking about gpio3.

Will have a look into it for you and post back

Gumboot13, 2020-08-09:
> Right. Where the hell is the GPIO4 then on a Shelly 1? :)

Believe I had gpio4 and gpio5 mixed up.

I know pins gpio3 and gpio5 are good to use at being held low on boot

morphix9, 2020-08-09:
Gpio5 is the switch input (see schematic above). It's not a pin like GPIO0, 1 or 3, where you cut put a wire on. Or am I wrong?

Gumboot13, 2020-08-09:
> Gpio5 is the switch input (see schematic above). It's not a pin like GPIO0, 1 or 3, where you cut put a wire on. Or am I wrong?

Correct, just wire it into the screw down terminal

morphix9, 2020-08-15:
I've tried with GPIO 5 but can't get it running (switched from GPIO1 to 5) :(

My config is:

`{"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2},"t":40,"d":52,"b":[],"f3":[{"g":5,"t":0}],"f4":[{"g":5}],"f2":[{"g":3,"t":0}],"f5":[{"g":3}],"s":1},{"1":{"s":[{"a":1}]},"i":0.5,"b":[],"s":0}]}`

Could you share yours @Gumboot13 ?

@fIctioN600 you are running it with GPIO 1 and 3 without any issues (on startup)?

fIctioN600, 2020-08-15:
I do. Never recognized problems with it.

Gumboot13, 2020-08-15:
> I've tried with GPIO 5 but can't get it running (switched from GPIO1 to 5) :(
> 
> Could you share yours @Gumboot13 ?
> 
I'm running Esphome on my devices. Happy to share that config if you want

Gumboot13, 2020-08-15:
> I do. Never recognized problems with it.

Have you tried holding gpio1 low on boot up?

morphix9, 2020-08-16:
I've tried booting it up with GPIO1 on low and it's not working (HAA not responding). @RavenSystem is this a Shelly 1 specific issue or HAA?

@Gumboot13 

Do you have one sensor on GPIO5 (SW) and one on GPIO3? Could the GPIO5 sensor use ground from the GND pin output or does it have to be 24DC-? Maybe stupid question, but I'm not an electrician ;P

Gumboot13, 2020-08-16:
> I've tried booting it up with GPIO1 on low and it's not working (HAA not responding). @RavenSystem is this a Shelly 1 specific issue or HAA?
>
> @Gumboot13
>
> Do you have one sensor on GPIO5 (SW) and one [...]

I use ground for both. 3 wires for two reed switches. Ground, gpio3 and gpio5

DenisG1994, 2022-07-04:
> Hi, I have a problem with a sensor connected to GPIO5. It simply doesn't work. Previously I had one sensor connected to GPIO1 and the other to GPIO3 but during the boot when GPIO1 was low, the device did not boot properly. This has been described above. So, I decided to use GPIO5 instead but I am unable to low it, I tried with L and GND pin. I'm using 12V DC. Can anyone help me? My JSON below:
> 
> {"c":{"b":[{"g":1,"t":5}]},"a":[{"t":40,"d":35,"c":49,"f3":[{"g":5,"t":0}],"f4":[{"g":5}],"f2":[{"g":3,"t":0}],"f5":[{"g":3}],"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2}}]}

After a LOT of time and test I finally managed to use 2 sensors (open on GPIO3 and close on GPIO5) using the JSON down there... The "problem" (at least in my case) was that GPIO5 is "active low" so I had to invert the input.
PS. I used the Json configurator so maybe there is some extra code that's not needed...

{"c":{"0":{"r":[{"g":4}]}},"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"r":[{"g":4,"v":1,"i":0.5}]},"2":{"r":[{"g":4,"v":1,"i":0.5}]},"3":{"r":[{"g":4,"v":1,"i":0.5}]},"t":40,"d":27,"s":5,"b":[],"f2":[{"g":3,"t":0,"i":0}],"f3":[{"g":5,"t":0,"i":1,"p":0}],"e":35,"f4":[{"g":5,"t":1,"p":0,"i":1}],"f5":[{"g":3,"t":1}]}]}