RavenSystem/esp-homekit-devices #819: SONOFF 4ch Pro R2 4 Water Valve (Sprinkler)

blackwalkerhd, 2020-02-26:
hi all!

my SONOFF 4ch Pro R2 work fine with the standard json string with 4 switch:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]}]}

when i use a string with 4 Water Valve (Sprinkler) , i see the 4 water valve switch and work fine (on/off) but i try to use them in a Homekit automation, when I have to install the accessory, it does not find it! I also tried to create a new room and move them there but nothing ... I tried two JSON strings that I attach:

First:

{"c":{"o":0,"l":13,"i":0,"f":10},"a":[{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":20,"b":[{"g":0,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"t":20,"b":[{"g":9,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":4,"v":0}]},"1":{"r":[{"g":4,"v":1}]},"t":20,"b":[{"g":10,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"t":20,"b":[{"g":14,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]}]}

Second:

{"c":{"o":0,"m":60,"l":13,"i":0,"f":10,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1}]},"1":{"r": [{"g":12,"v":0}]},"t":20,"b":[{"g":0,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":5,"v":1}]},"1": {"r":[{"g":5,"v":0}]},"t":20,"b":[{"g":9,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":4,"v":1}]},"1": {"r":[{"g":4,"v":0}]},"t":20,"b":[{"g":10,"t":1,"p":1}],"s":0,"w":0,"d":0,"f0":[]},{"0":{"r": [{"g":15,"v":1}]},"1":{"r":[{"g":16,"v":0}]},"t":20,"b":[{"g":14,"t":1,"p":1}],"s":0,"w":0,"d":0,"f0":[]}]}

all show correctly the icon and work on/off but nothing more...

someone PLS?
thanks

teymur1988, 2020-02-27:
Hello, I've had the same thing happened to me, Homekit doesn't see the water valve in automations. Create an automation in the Eve app and you will see the water valve there.

pilot1981, 2020-03-23:
I solved in this way:
1) create automation timer in EVE APP, here you have to create new scene
2) correct starting date and time in Apple home app (here you can find new scene and automation)

This work fine for me.

closed 2020-04-22 by stale[bot]

KrzysiekRadek, 2020-05-09:
Hi All :)
I need a sprinkler string for Sonoff BASIC ...
can you help me?

mat130780, 2020-06-12:
Hi,

I am currently trying to set up a 4ch pro r2 as a sprinkler controller. Is it possible to receive a notification from the Home app when a sprinkler is activated or deactivated?

pilot1981, 2020-06-12:
No

pilot1981, 2020-06-12:
You can try Pushcut app, but it worked for me only first time I configured it!