RavenSystem/esp-homekit-devices #823: IR Codes Collection

RavenSystem, 2020-02-28:
Users can post here working IR codes to be included in the IR Codes database:
https://github.com/RavenSystem/esp-homekit-devices/wiki/IR-Codes-Database

Template:
- Vendor: 
- Type: TV, HVAC, Fan, Music Player...
- Remote model: 
- Protocol:
- Codes:

GPL71, 2020-02-28:
Vendor: PHILIPS
Type: TV
Remote model: RC6 20 bit frequency 36Khz
Protocol: FiBK00AFAF00AF
Codes:
            On / Off    : AbAaAaCcAaAaAaAaAaAaAaAaAaAaAaBaAbAa          
            Channel +: AbAaAaAbBaAaAaAaAaAaAaAaAaAaBbAaAaAaA
            Channel - : AbAaAaCcAaAaAaAaAaAaAaAaAaCbAaAaAa
            Volume  + : AbAaAaAbBaAaAaAaAaAaAaAaAaAaAaCbAaAaAa
            Volume  - : AbAaAaCcAaAaAaAaAaAaAaAaAaAaCbAaAa 
            Up            : AbAaAaAbBaAaAaAaAaAaAaAaAaBbBaAbAaAa
            Down       : AbAaAaCcAaAaAaAaAaAaAaAaBbBaAbAa
            Right       : AbAaAaAbBaAaAaAaAaAaAaAaAaCbBaAbBa
            Left         : AbAaAaCcAaAaAaAaAaAaAaAaBbBaAbBb
            Mute       : AbAaAaAbBaAaAaAaAaAaAaAaAaAaAaAaBaAb 
            Menu      :

GPL71, 2020-02-29:
Vendor: PHILIPS
Type: TV
Remote model: RC6 20 bit frequency 36Khz
Protocol: FiBK00AFAF00AF
Codes:
Menu : AbAaAaAbBaAaAaAaAaAaAaAaAaBbBbBbAa
Input  : AbAaAaCcAaAaAaAaAaAaAaAaAaBaAaAbAaAa
Last   : AbAaAaCcAaAaAaAaAaAaAaAaAaAaAaBbBb
OK     : Looking for

Partizan7676, 2020-03-01:
here - http://www.remotecentral.com/ you can find a lot of codes. use it when looking for power on/off discrete codes.

GPL71, 2020-03-04:
Vendor: PHILIPS
Type: TV
Remote model: RC6 20 bit frequency 36Khz
Protocol: FiBK00AFAF00AF
JSON for follow commands: ON, OFF, ArrowUP, ArrowDOWN, ArrowLEFT, ArrowRIGHT, OK.
NOTE: 
The json include IR and Http commands. 
Replace IP address in json with your TV IP address.
Let me know if this jSON works fine.
Thank you to @RavenSystem for helping and this great work!

{"c":{"t":4,"x":35,"p":"fcBK00AFAF00AF"},"a":[{"t":60,"q0":[{"h":"192.168.0.37","r":0}],"q1":[{"h":"192.168.0.37"}],"i":[{"n":"HDMI 1","0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 1"}"}]}},{"n":"HDMI 2","0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 2"}"}]}},{"n":"HDMI 3","0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 3"}"}]}},{"n":"HDMI 4","0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 4"}"}]}}],"0":{"r":2,"i":[{"c":"AbAaAaCcAaAaAaAaAaAaAaAaAaAaAaBaAbAa"}]},"1":{"r":2,"i":[{"c":"AbAaAaCcAaAaAaAaAaAaAaAaAaAaAaBaAbAa"}]},"6":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorUp"}"}]},"7":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorDown"}"}]},"8":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorLeft"}"}]},"9":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorRight"}"}]},"10":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Confirm"}"}]}}]}

TriderG75, 2020-03-24:
Vendor: SAMSUNG
Type: TV
Remote model: AA59-00622A 38KHz
Protocol: J*J*AaAaAaDFAa
Codes:

Tools-	CeCeBaAbAcAaBaA
CH List-	CeCeBaAaBcAaAbA
Info-		CeCeEhC
Hdmi-	CeCeBaAcAbAaCa
1-		CeCgAeBbE
2-		CeCeAaAfAaE
3-		CeCfBeAbE
4-		CeChAdCaD
5-		CeCeAbAeBaD
6-		CeCfAaAdAaAaD
7-		CeCgBdBbD
8-		CeCeAaBeAbD
9-		CeCfCdAcD
0-		CeCeAcAdCaC

wayn3, 2020-05-22:
Vendor: Microsoft
Type: Xbox One
Device Code: 128
Subdevice Code 216
Protocol: NEC Extended
Frequency 38Khz
HAA Protocol: U5J(AcAcAcDFAc
Command   No | Command Code | Binary | HAA | Function
-- | -- | -- | -- | --
41 | 00101001 | 00000001000110111001010001101011 | gAcBaCbAaAcBaAaB | Off
42 | 00101010 | 00000001000110110101010010101011 | gAcBaBaAaAaAbAaAaAaB | On
30 | 00011110 | 00000001000110110111100010000111 | gAcBaBaDcAdC | Up
31 | 00011111 | 00000001000110111111100000000111 | gAcBaGhC | Down
32 | 00100000 | 00000001000110110000010011111011 | gAcBaBeAbEaB | Left
33 | 00100001 | 00000001000110111000010001111011 | gAcBaCdAcDaB | Right
34 | 00100010 | 00000001000110110100010010111011 | gAcBaBaAcAbAaCaB | Ok
11 | 00001011 | 00000001000110111101000000101111 | gAcBaDaAfAaD | Enter
35 | 00100011 | 00000001000110111100010000111011 | gAcBaDcAdCaB | Exit
22 | 00010110 | 00000001000110110110100010010111 | gAcBaBaBaAcAbAaC | Play
24 | 00011000 | 00000001000110110001100011100111 | gAcBaBcBcCbC | Pause
14 | 00001110 | 00000001000110110111000010001111 | gAcBaBaCdAcD | Mute
16 | 00010000 | 00000001000110110000100011110111 | gAcBaBdAcDaC | Vol+
17 | 00010001 | 00000001000110111000100001110111 | gAcBaCcAdCaC | Vol-
38 | 00100110 | 00000001000110110110010010011011 | gAcBaBaBbAbAbBaB | Xbox
40 | 00101000 | 00000001000110110001010011101011 | gAcBaBcAaAbCaAaB | Eject
101 | 01100101 | 00000001000110111010011001011001 | gAcBaCaAbBbAaBbA | B
102 | 01100110 | 00000001000110110110011010011001 | gAcBaBaBbBaAbBbA | A
103 | 01100111 | 00000001000110111110011000011001 | gAcBaEbBdBbA | Y
104 | 01101000 | 00000001000110110001011011101001 | gAcBaBcAaBaCaAbA | X

I have also generated all possible commands but most have no use

wayn3, 2020-05-22:
Vendor: Sony
Type: TV
Device Code: 1
Protocol: SIRC 12 bit
Frequency 40Khz
HAA Protocol: E@AkAkAkB*Ak00
Command   No | Command Code | Binary | HAA | Function
-- | -- | -- | -- | --
47 | 0101111 | 111101010000 | DaAaAd | Power Off
46 | 0101110 | 011101010000 | aCaAaAd | Power On
116 | 1110100 | 001011110000 | bAaDd | Up
117 | 1110101 | 101011110000 | AaAaDd | Down
52 | 0110100 | 001011010000 | bAaBaAd | Left
51 | 0110011 | 110011010000 | BbBaAd | Right
101 | 1100101 | 101001110000 | AaAbCd | Enter/Ok
99 | 1100011 | 110001110000 | BcCd | Exit/Back
20 | 0010100 | 001010010000 | bAaAbAd | Mute/Unmute
18 | 0010010 | 010010010000 | aAbAbAd | Volume Up
19 | 0010011 | 110010010000 | BbAbAd | Volume Down
96 | 1100000 | 000001110000 | eCd | Menu
64 | 1000000 | 000000110000 | fBd | Video 1
65 | 1000001 | 100000110000 | AeBd | Video 2
66 | 1000010 | 010000110000 | aAdBd | Video 3
68 | 1000100 | 001000110000 | bAcBd | PC
71 | 1000111 | 111000110000 | CcBd | Video 4/Component 1
72 | 1001000 | 000100110000 | cAbBd | Video 5/Component 2
73 | 1001001 | 100100110000 | AbAbBd | Video 6/HDMI 1
74 | 1001010 | 010100110000 | aAaAbBd | Video 7
75 | 1001011 | 110100110000 | BaAbBd | Video 8
76 | 1001100 | 001100110000 | bBbBd | Video 9
77 | 1001101 | 101100110000 | AaBbBd | Video 10
78 | 1001110 | 011100110000 | aCbBd | Video 11
79 | 1001111 | 111100110000 | DbBd | Video 12

I have also generated all possible commands but most have no use, I have included additional codes likely to work **but** I have an older TV that has a bunch of discrete codes for components, rca port that I would never use and only one discrete code for HDMI, newer TV's would likely use these other codes

pawelosin, 2020-06-13:
> Vendor: PHILIPS
> Type: TV
> Remote model: RC6 20 bit frequency 36Khz
> Protocol: FiBK00AFAF00AF
> JSON for follow commands: ON, OFF, ArrowUP, ArrowDOWN, ArrowLEFT, ArrowRIGHT, OK, ack, Play/Pause, Info, Mute, Volume, Settings and HDMI sources.
> NOTE:
> The json include IR and Http commands.
> Replace IP address in json with your TV IP address.
> Let me know if this jSON works fine.
> Thank you to @RavenSystem for helping and this great work!
> 
> {"c":{"t":4,"x":35,"p":"fcBK00AFAF00AF"},"a":[{"0":{"r":2,"i":[{"c":"AbAaAaCcAaAaAaAaAaAaAaAaAaAaAaBaAbAa"}]},"1":{"r":2,"i":[{"c":"AbAaAaCcAaAaAaAaAaAaAaAaAaAaAaBaAbAa"}]},"6":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorUp"}"}]},"7":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorDown"}"}]},"8":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorLeft"}"}]},"9":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorRight"}"}]},"10":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Confirm"}"}]},"11":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Back"}"}]},"13":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"PlayPause"}"}]},"15":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Info"}"}]},"20":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Mute"}"}]},"21":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Mute"}"}]},"22":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"VolumeDown"}"}]},"23":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"VolumeUp"}"}]},"30":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Options"}"}]},"t":60,"q0":[{"h":"192.168.0.37","r":0}],"q1":[{"h":"192.168.0.37"}],"i":[{"0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 1"}"}]},"n":"HDMI 1"},{"0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 2"}"}]},"n":"HDMI 2"},{"0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 3"}"}]},"n":"HDMI 3"},{"0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 4"}"}]},"n":"HDMI 4"}]}]}

> Vendor: PHILIPS
> Type: TV
> Remote model: RC6 20 bit frequency 36Khz
> Protocol: FiBK00AFAF00AF
> JSON for follow commands: ON, OFF, ArrowUP, ArrowDOWN, ArrowLEFT, ArrowRIGHT, OK, ack, Play/Pause, Info, Mute, Volume, Settings and HDMI sources.
> NOTE:
> The json include IR and Http commands.
> Replace IP address in json with your TV IP address.
> Let me know if this jSON works fine.
> Thank you to @RavenSystem for helping and this great work!
> 
> {"c":{"t":4,"x":35,"p":"fcBK00AFAF00AF"},"a":[{"0":{"r":2,"i":[{"c":"AbAaAaCcAaAaAaAaAaAaAaAaAaAaAaBaAbAa"}]},"1":{"r":2,"i":[{"c":"AbAaAaCcAaAaAaAaAaAaAaAaAaAaAaBaAbAa"}]},"6":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorUp"}"}]},"7":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorDown"}"}]},"8":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorLeft"}"}]},"9":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"CursorRight"}"}]},"10":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Confirm"}"}]},"11":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Back"}"}]},"13":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"PlayPause"}"}]},"15":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Info"}"}]},"20":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Mute"}"}]},"21":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Mute"}"}]},"22":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"VolumeDown"}"}]},"23":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"VolumeUp"}"}]},"30":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/input/key","c":"{"key:":"Options"}"}]},"t":60,"q0":[{"h":"192.168.0.37","r":0}],"q1":[{"h":"192.168.0.37"}],"i":[{"0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 1"}"}]},"n":"HDMI 1"},{"0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 2"}"}]},"n":"HDMI 2"},{"0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 3"}"}]},"n":"HDMI 3"},{"0":{"h":[{"h":"192.168.0.37","p":1925,"m":2,"u":"1/sources/current","c":"{"id:":"HDMI 4"}"}]},"n":"HDMI 4"}]}]}

I have tried your config code many times, but I can't configure my device. After paste the JSON to my device, the device it is still in setup mode. I have checked the json in JSONLint - The JSON Validator, and result was 

"Error: Parse error on line 26:...","c": "{"key: ":"CursorU
---------------------^
Expecting 'EOF', '}', ':', ',', ']', got 'undefined'
Any idea?

GPL71, 2020-06-22:
> I have tried your config code many times, but I can't configure my device. After paste the JSON to my device, the device it is still in setup mode. I have checked the json in JSONLint - The JSON [...]

Please open a new specific ISSUE. This is a collection.
Unfortunatly my TV broken and I couldn't continue test with Json.
However you could use HAA raw commands in the beginning to test.

wayn3, 2020-07-30:
Vendor: Thermofilm
Type: Heatstrip Controller [TT-MTM](https://heatstrip.com.au/product/heatstrip-hardwired-controller-tt-mtm/)
Device Code: 65
Subdevice Code: 234
Protocol: NEC Extended
Frequency 38Khz
HAA Protocol: U5J(AcAcAcDFAc
Command   No | Command Code | Binary | HAA | Function
-- | -- | -- | -- | --
2 | 00000010 | 10000010010101110100000010111111 | AeAbAaAaCaAfAaF | Heat Level (Non Discrete)
6 | 00000110 | 10000010010101110110000010011111 | AeAbAaAaCaBeAbE | Power Toggle
27 | 00011011 | 10000010010101111101100000100111 | AeAbAaAaEaBeAbC | Timer (Non Discrete)

I have also generated all possible commands in search of discrete codes and will update if I find anything.

wayn3, 2020-09-23:
🛠️

Kristian8606, 2020-11-04:
Vendor: Philips
Type: TV
Model: 43PUS6754/12

`{
  "c":{ "t":14 },
  "a":[{
    "t":60,
    "xa":0,
    "x":0,
    "i":[
      { "n":"MENU", "0":{ "i": [{ "w":"FmBFALBGALAAALAACWCRAGAAALAAALA0ALA0ALAAALAAALAAALABBRBIBQBIBRBIALAAAL" }] } },
      { "n":"SOURCE", "0":{ "i":[{ "w":"FmBFALBGALAAALAACWCRAGAAALAAALAAALA0ALAAALAAALAAALABALAABQACAL0>AMBGALAAALAAAL" }] } },
      { "n":"MUTE", "0":{ "i":[{ "w":"FmBFAHBLAGAFAGAFCRCWABAFAGAEAGAFAGAEAGAFAGAEAGAFAGAGAGAFAGAFAGAFBLAGAHBKBM" }] } },
      { "n":"AMBILIGHT", "0":{ "i":[{ "w":"FmBFALBGALAAALAAALBCBRABALA0ALA0ALA0ALA0ALAAALAAALAABRBJAL0>ANAABRABALAAALAAAL" }] } }
    ],
    "0":{"i":[{"w":"FmBFALBGALA0ALA0ALBCBRABAMA0ALAAALAAALAAALA0ALAAALA0ALAAALAAALA0ALAABRABALBGALAAAL"}] },
    "1":{"i":[{"w":"FmBFALBGALA0ALA0ALBCBRABAMA0ALAAALAAALAAALA0ALAAALA0ALAAALAAALA0ALAABRABALBGALAAAL"}] },
    "6":{"i":[{"w":"FmBFALBGALAAALAAALBDBRABALAAALAAALAAALAAALAAALAAALAAALABBRBIBQACALBGALA0ALAAAL"}] },
    "7":{"i":[{"w":"FmBEAMBGALAAAL0>ANBDBRABALA0ALAAALAAALAAALAAALA0ALA0ALABBRBHBRABALBGALAABR"}] },
    "8":{ "i":[{"w":"FmBFALBGALAAALAACWCSAGAAALAAALAAALAAALAAALAAALAAALABBRBIBRABALBGBRBIAL" }] },
    "9":{ "i":[{ "w":"FmBFALBGALAAALAACXCRAGA0ALAAALAAALAAALAAALA0ALA0ALABBRBIBRABALBGBRABAL" }] },
    "10":{ "i":[{ "w":"FmBFALBGALAAALAACWCRAGAAALAAALA0ALAAALAAALAAALA0ALABBRBIBQACALAAALBGALAAAL" }] },
    "11":{ "i":[{ "w":"FmBFALBGALAAALA0CWCSAGAAALAAALAAALAAALAAALAAALAAALABALAAALAAALA0BRBIBQBIAL" }] },
    "13":{ "i":[{ "w":"FmBFALBGALAAALA0CWCRAGAAALAAALAAALAAALAAALAAALAAALABALAABRBIBRABALBGALAAAL" }] },
    "17":{"i":[{"w":"FmBFALBGALAAALAACWCRAGA0ALAAALAAALAAALAAALA0ALAAALABALAAALAAALAABRABALAAALAAAL"}]},
    "22":{"i":[{"w":"FmBEALBGALA0ALA0ALBCBRABALA0ALA0ALA0AMA0ALA0ALA0AMA0AMAAAMA0ALA0BRBHALA0ALA0ALA0AL"}] },
    "23":{"i":[{"w":"FmBEALBGALA0AMA0CXCRAHA0ALA0ALA0AMA0AMA0ALA0ALAAALAAALAAALA0BRBHAMA0ALA0BR"}] },
    "30":{"i":[{"w":"FmBFALBGALAAALAAALBCBRABALA0ALA0ALA0ALAAALAAALA0ALA0ALAABRBIALAAALA0ALAAALAAALA0AL"}] },
    "q0":[{"h":"192.168.0.122","r":0}],
    "q1":[{"h":"192.168.0.122","r":1}]
  },
  {
    "i":0.5,
    "1":{ "s":[{ "a":1 }]}
  }]
}`

Trx899, 2020-11-15:
Vendor: Irobot (Roomba)
Type: Vacuum Robot
Remote model: Standard IR remote for 500 to 800 Series
Frequency: 38 Khz
Protocol: RAW
Codes:
clean
F=BXBXF=BXF=BXF=F=BXBXF=BXF=BXs@F=BXBXF=BXF=BXF=F=BXBXF=BXF=BX
spot
F=BXBXF=BXF=BXF=BXF=F=BXBXF=BXs@F=BXBXF=BXF=BXF=BXF=F=BXBXF=BX
dock
F=BXBXF=BXF=BXF=BXF=F=BXBXF=F=s@F=BXBXF=BXF=BXF=BXF=F=BXBXF=F=
Up
F=BXBXF=BXF=BXF=BXF=BXF=F=BXBXs@F=BXBXF=BXF=BXF=BXF=BXF=F=BXBX
LEFT
F=BXBXF=BXF=BXF=BXF=BXF=F=BXF=s@F=BXBXF=BXF=BXF=BXF=BXF=F=BXF=
RIGHT
F=BXBXF=BXF=BXF=BXF=BXF=BXF=F=s@F=BXBXF=BXF=BXF=BXF=BXF=BXF=F=
OK
F=BXBXF=BXF=BXF=BXF=BXF=BXF=F=s@F=BXBXF=BXF=BXF=BXF=BXF=BXF=F=

okanduzyel, 2021-03-04:
Vendor: Mitsubishi 
Type: TV, HVAC, Fan, Music Player...
Remote model: E12E79426 
Frequency:38khz
Protocol: MITSUBISHI112
Codes:

https://github.com/RavenSystem/esp-homekit-devices/wiki/Thermostat

```
Protocol:			HlDDAQAEAQB>AK
0:				BcAbBaAbBaBbAbAtAbAgAhAaCzhAcCa
1:				BcAbBaAbBaBbAbAqAbAbAgDeAaCzhBeA
2:				BcAbBaAbBaBbAbAqAbAbBoAaCziBaCa
3:				BcAbBaAbBaBbAbAqAbAbApAaDzjAaDa
4:				BcAbBaAbBaBbAbAqAbAbBfDeAeDzhAaAD
```

closed 2021-08-22 by RavenSystem