RavenSystem/esp-homekit-devices #824: Unable to make device operational

Quoshak, 2020-02-29:
I have an issues when flashing SONOFF Basic (also Sonoff S26)

Procedure:
1. flash SONOFF with esptools.py - OK
2. connect to device WiFi (HAA-XXXXXX) - OK
3. enter JSON string, select Automatic OTA updates checkbox, select network (with entered password) then click save - OK

After some time I would expect to be able to add a device to HomeKit, but am not. I am still able to access setup mode, this time via IP on previously selected network from a browser.

See logs in attachment...
[screenlog.log](https://github.com/RavenSystem/esp-homekit-devices/files/4270288/screenlog.log)

Details:
- macOS
- AirPort Extreme router
- latest successfully added device has RavenSystem HAA 1.3.1

claire078, 2020-02-29:
I think HAA server doesn’t work. I can’t update OTA my devices to latest firmware and can’t set a new device, like you.

RavenSystem, 2020-02-29:
GitHub changed some parameters (without any reason) and now OTA installations are broken.

RavenSystem, 2020-02-29:
https://github.com/RavenSystem/esp-homekit-devices/issues/825

closed 2020-02-29 by RavenSystem