RavenSystem/esp-homekit-devices #825: OTA Installations are broken

RavenSystem, 2020-02-29:
GitHub servers, used to install firmware by OTA procedure, did some changes and now OTA installations are broken. Keep calm, and stay tuned to know how to scape from this hole.

RavenSystem, 2020-02-29:
https://github.community/t5/GitHub-API-Development-and/GitHub-changed-the-capitalisation-of-the-HTTP-headers-and-OTA/m-p/48247

HomeACcessoryKid, 2020-02-29:
If everyone follows this link and kudos this request, it might actually help convince the GitHub team...

GPL71, 2020-02-29:
@RavenSystem @HomeACcessoryKid 
Thank you ... just do it

RavenSystem, 2020-02-29:
https://github.com/HomeACcessoryKid/life-cycle-manager/pull/22

RavenSystem, 2020-02-29:
New installations have been fixed and they will work. Ensure that you are using the last haaboot.bin

gnoto, 2020-03-01:
do I have to make a re-flash of all the accessories?

archiolochos, 2020-03-03:
> 
> 
> New installations have been fixed and they will work. Ensure that you are using the last haaboot.bin

How do i change the haaboot.bin on a 1.7-Device? is ther a way without FTDI-Flash?

sblitzer, 2020-03-03:
@RavenSystem can you update us please?
Do we need to flash all devices physically ? Or u will release update?

HomeACcessoryKid, 2020-03-03:
All,

There is a request outstanding with the GitHub support team.

IF they roll back the change they did on the HTTP header “Location:” then all will work again and you can download the next version of haaboot or lcm.

So, I recommend some patience since this kind of organisations do not act in a 2 workdays notice.

I still have a reasonable hope this will be fixed in the next two weeks.

Go to the page https://github.community/t5/GitHub-API-Development-and/GitHub-changed-the-capitalisation-of-the-HTTP-headers-and-OTA/td-p/48247 and give a kudos to the repair request.

Meanwhile, cross your fingers,
my 2 cents,
HacK

tonysprenk, 2020-03-03:
We are still waiting on github to reverse the change (temporarily). Until then we have to wait. Flashing by wire is always an option but it will mean all HomeKit ID’s will be reset. There is a way to do it without resetting HomeKit ID and @RavenSystem has informed us in the discord that he will make a step by step of how to do so if github is not willing to reverse the change. For now just don’t panic and wait patiently 😊.

i3laze, 2020-03-15:
I’m planning on flashing a new device.

@RavenSystem, how about you release a new HAA OTA version, which tries handling both old and new HTTP Headers?
Who knows how much time GitHub can stall on this issue.

Kienz, 2020-03-15:
@i3laze For new installations its been fixed.
> New installations have been fixed and they will work. Ensure that you are using the last haaboot.bin

morphix9, 2020-03-18:
@RavenSystem so what's the way out of this? Will you release instructions to flash the OTA without resetting the HomeKit id/settings? What should we do?

HomeACcessoryKid, 2020-03-18:
I have updated my LCM with several features, including an emergency mode.  
This can download a new otaboot from any http server.  
Since it verifies the signature it is still protected against hijacking the device.  
I suggest HAA gets updated with these features as well…

I for one have tested my new LCM to work by uploading only to sector 0x2000 and this preserves the HomeKit settings.

macjl, 2020-03-23:
Is there a way to fake the GitHub servers to update to the last firmware?
Is the update process checking the validity of the GitHub certificate when updating?

HomeACcessoryKid, 2020-03-24:
Yes, the entire point of LCM is to protect against malware injection. Afterward the emergency mode should have been there from the beginning, but this is how we learn from our mistakes.

Kienz, 2020-03-24:
> I have updated my LCM with several features, including an emergency mode. This can download a new otaboot from any http server. Since it verifies the signature it is still protected against [...]

@HomeACcessoryKid Do you have an example?

HomeACcessoryKid, 2020-03-24:
esptool.py -p /dev/cu.usbserial-* --baud 230400 write_flash -fs 8m -fm dout -ff 40m \
        0x2000 versions/1.9.5v/otaboot.bin

HomeACcessoryKid, 2020-03-24:
Better do this, it then preserves the bootflags as well:

esptool.py -p /dev/cu.usbserial-* write_flash  0x2000 versions/1.9.5v/otaboot.bin

morphix9, 2020-03-24:
But this can't be done if you have the HAA OTA currently installed, right?

HomeACcessoryKid, 2020-03-24:
no, the example was given for LCM otaboot  
I for one do not use HAA ota (yet)

morphix9, 2020-03-24:
@RavenSystem are you planning to do an update on the OTA? Can you share some info?

xrust83, 2020-03-31:
I managed to update without a reset.
I use sonoff basiс + Si7021 but the firmware gives too much accuracy for the sensor 22.2000007629395 °C temp and 53.5000006543235 % humidity. Can you round to 0.1?

hillaliy, 2020-04-04:
share your secret

RavenSystem, 2020-04-04:
> share your secret

@hillaliy read first comment.

claire078, 2020-04-04:
> I managed to update without a reset.

it would be better, update without flash. 😉

hillaliy, 2020-04-04:
I don't understand,
How can I update without flash mode

peros550, 2020-04-05:
Guys, the link found in the original post (https://github.community/t5/GitHub-API-Development-and/GitHub-changed-the-capitalisation-of-the-HTTP-headers-and-OTA/m-p/48247) no longer works. 

Any progress with Github?

tonysprenk, 2020-04-05:
Wow that’s strange. Just a few hours ago I contacted github support asking them what the status was of this issue. I referenced the above link in my request. I verified that the link was operational. Now it seems the issue has been removed from the support pages.

HomeACcessoryKid, 2020-04-05:
I had made post with a "vote" to see if people were in favour of making a fuz or giving up.  
When I updated the main post with a link to post#15 the article was marked as spam

I have used the feedback form for this, indicating it is not spam. And asking for a reply, any reply.
If we are lucky, on Monday they might un-spam it.
Anyway, LCM version 2.0.0 is almost ready and Jose is looking at a backup server mode...
So either way, progress might be made...

tonysprenk, 2020-04-05:
> I had made post with a "vote" to see if people were in favour of making a fuz or giving up.
>
> When I updated the main post with a link to post#15 the article was marked as spam
>
>
>
> I have used the [...]

Thanks allot for the clarification! As I understand it the solutions you mention are only for newly flashed devices right? Not for devices that are already flashed OTA?

HomeACcessoryKid, 2020-04-05:
Correct. Unless GitHub changes that header, all those devices cannot be updated without serial flashing. I those cases, if you use HAA, you can also switch to HAA_OTA if you estimate it to be sufficiently able to protect you in the future for ever having to go through this again.
I still have to evaluate HAA_OTA so cannot tell if it is up to it already.

peros550, 2020-04-05:
@HomeACcessoryKid Supposing Github will not help, isn't any other way to save those devices running already a previous version of LCM? I have a few smart bulbs which were originally flashed with Tuya convert (OTA) hence having to flash by wire is quite difficult

HomeACcessoryKid, 2020-04-05:
If there would have been the remotest way of achieving anything, we would have found it by now.
Key observation is that your device is a state machine that acts on what it gets.
Since you cannot ADD anything to and it is therefor driving to a crash, there is no way around.

seritos, 2020-04-06:
> I had made post with a "vote" to see if people were in favour of making a fuz or giving up.
> When I updated the main post with a link to post#15 the article was marked as spam
>
> I have used the [...]

Keep us updated plz. 

Thnx!

HomeACcessoryKid, 2020-04-12:
LCM 2.0.1 is now the latest release. The readme has instructions to upgrade using solder iron.
No news from GitHub

HomeACcessoryKid, 2020-04-13:
Tried to upgrade a SonoffBasic from RavenCore 0.5.15 by using LCM 2.0.1 and it turns out that HAA overwrites LCM otamain code by using a very big 8 sector sysparam definition.

Considering that HAA_OTA still doesn't have an emergency mode, I do not want to switch to HAA_OTA, at least not till that is done

What should be do here?
Maybe fork HAA and bring sysparam back to normal sized 2 sectors which will not clobber LCM?

pilot1981, 2020-04-13:
IMPORTANT UPDATE:
if you upgrade from RC or HAA version without dummy switch to go in setup mode, this procedure create itself dummy switch!!! It isn't necessary to do any reset of Homekit ID, so you haven't to touch/modify any existing homekit accessory and automation; you have only to move dummy switch to righe room...

this is very useful to speed up migration ta vastest version of HAA with working OTA features!!!!

have a good migration task!

peros550, 2020-04-13:
> Tried to upgrade a SonoffBasic from RavenCore 0.5.15 by using LCM 2.0.1 and it turns out that HAA overwrites LCM otamain code by using a very big 8 sector sysparam definition.
>
> Considering that [...]

Am I get it right? Newest versions of HAA are not compatible with LCM v2? 
If yes, that would be a big drawback for me because I liked LCM new features (emergency mode & repo reset tool)

HomeACcessoryKid, 2020-04-13:
Jose and i are working on that

BR,
HacK

peros550, 2020-04-13:
Thank you guys! I think it's good think that Github hasn't fixed their issue yet.

pilot1981, 2020-04-13:
@peros550 you can use dummy switch to enter in setup mode?!

peros550, 2020-04-14:
Hi @pilot1981 

I have a few HAA devices running LCM old version and HAA 1.6.4. I can always go into setup mode using either the emergency mode or switching on-off (8x). 

I am not fun of the idea of having an additional dummy switch, unless there is a good reason for that....?

morphix9, 2020-04-14:
I would find those virtual switches quite useful for some devices, but can't find something in the wiki on that. Are those possible with HAA or was this only possible with the old RavenCore?

Shimakuchi, 2020-04-16:
I have no Idea how far your security is implemented but ... is it maybe possible to modify the headers with a proxy?

"[Requestly](https://chrome.google.com/webstore/detail/requestly-redirect-url-mo/mdnleldcmiljblolnjhpnblkcekpdkpa)" or "[Charles](https://inviqa.com/blog/guide-manipulating-http-charles-proxy)" for example can do such things.

Really hope there will be ANY other solution than flash the devices over the pin headers again 🤦‍♂️

macjl, 2020-04-16:
I don't know if you can tell LCM to use an http proxy. Is there a way to do that?

uDuCkV, 2020-04-16:
A transparent proxy maybe? Is it encrypted or unencrypted traffic?

macjl, 2020-04-16:
Encrypted. That's why transparent proxy won't be able to change headers.

HomeACcessoryKid, 2020-04-16:
and not only encrypted but actively checking the server certificates!

tonysprenk, 2020-04-16:
If we could fake the github server than anyone could and none of our devices would be safe. Maybe it’s possible to create a proxy that has the right URL that the HAA device can connect to which can relay the information from the actual github server. Not sure if that’s even physically possible with the certificates etc. And again it would mean that anyone could use that method to gain access to all our devices.

peros550, 2020-04-16:
@HomeACcessoryKid  In general LCM check validity of certificates, which is a highly secure thing to do, but haven't you designed it in a way to trust certificates signed by you?

If yes, maybe we can fix this

HomeACcessoryKid, 2020-04-18:
The 1.0.0 version did not contemplate the scenario where GitHub would make http headers incompatible. It does handle if GitHub will update their certificates (bound to happen one day, e.g. if they change their certificate provider). But without the basics of http working, it is broken and no way around it (unless you want to break GitHubs certificates, which would be breaking news around the world.
Anyhow, only with version 1.9.1+ will you be able to use GitHub again. And versions 2.0.0+ have an emergency mode that can save a device without needing GitHub.

HomeACcessoryKid, 2020-04-18:
> Am I get it right? Newest versions of HAA are not compatible with LCM v2?
If yes, that would be a big drawback for me because I liked LCM new features (emergency mode & repo reset tool)

The version 2.0.2 uses a special version created by @RavenSystem  that is compatible with LCM.  
@peros550, although already tested by myself, please check it out?

peros550, 2020-04-18:
@RavenSystem @HomeACcessoryKid Many thanks for your works guys. 

I just flashed the [2.0.2](https://github.com/HomeACcessoryKid/life-cycle-manager/releases/download/2.0.2/otamain.bin) otaboot.bin and watched the magic happening. LCM updated to latest version, then rboot, then HAA updated. 

The Sonoff S20 kept Homekit settings. As soon as the process completed, the accessory was available again in Home app. 

`But without the basics of http working, it is broken and no way around it`
@HomeACcessoryKid I'm not a security expert and don't try to pretend one. I was trying to think how we could alter http headers and came across  one tool, it's called mitmproxy. This tool is designed to intercept http/https and also has an API to make changes on the fly [[1]](https://mitmproxy.readthedocs.io/en/v2.0.2/features/replacements.html) [[2](https://docs.mitmproxy.org/stable/addons-events/)] and alter the http responses in the way we want. 

I actually followed this [guide ](https://www.dinofizzotti.com/blog/2019-01-09-running-a-man-in-the-middle-proxy-on-a-raspberry-pi-3/)to install it on a raspberry and do some tests. 

Assuming the http header issue can be solved, the next problem with this approach is trust. As you said, we can't brake security of Github certificates. On the other hand, this tools seems to provide some options for using Custom certificates, or CAs in the [Certificates section](https://docs.mitmproxy.org/stable/concepts-certificates/)

Do you think it's possible to setup a system like that and try to OTA devices that can't be flashed using soldering method?

i3laze, 2020-04-30:
Unless the device trusts MitM certificate representing the GitHub, there’s no luck faking it.
What happened with support ticket at Github portal? Any news from the support?

tonysprenk, 2020-05-11:
Just received this message from support: 

Apparently, the GitHub Community Forum entry has since been deleted. As such I have no insight into what exactly changed which broke this project's update procedure. As far as I can tell, they seem to mention a URL change of some sort.

I'm not sure if the following is applicable to the current situation, but some projects rely on undocumented URLs or query data in a way that is not via GitHub's API. In those cases, we can't provide support for any problems they might encounter.

API changes are usually communicated well ahead of time and come with deprecation warnings, unless an endpoint is part of a preview phase for which there's always a disclaimer that syntax and behavior can change without prior notice.
Let me know if you have any further questions.

i3laze, 2020-05-11:
Google cache has no copies of the original issue.
However, [Yandex does](https://yandexwebcache.net/yandbtm?lang=en&fmode=inject&tm=1589194334&tld=ru&la=1585495168&text=https%3A%2F%2Fgithub.community%2Ft5%2FGitHub-API-Development-and%2FGitHub-changed-the-capitalisation-of-the-HTTP-headers-and-OTA%2Fm-p%2F48247&url=https%3A%2F%2Fgithub.community%2Ft5%2FGitHub-API-Development-and%2FGitHub-changed-the-capitalisation-of-the-HTTP-headers-and-OTA%2Fm-p%2F50878&l10n=ru&mime=html&sign=ecb4fb4d126a1e8472d868ed7517b636&keyno=0 ) :)

I have a few RC devices left not flashed to HAA, however any resolution from GitHub is still welcome.

@HomeACcessoryKid, @tonysprenk, 
Please, use text from the cache URL above to reproduce the ticket.

HomeACcessoryKid, 2020-05-11:
or, you can use the ticket number of the original request for support.

Thank you for contacting GitHub Developer Support. We wanted to let you know that we've received your message and will get to it as quickly as possible.

Ticket ID: 582776

And the issue is not with the API but with the HTTP of the server, so that is below this API.
Although the RFC of HTTP allows using any capitalisation, the point is that changing this serves no use but creates this kind of nuisance

Below the text of the original support ticket request of 29 Feb 2020

Since today GitHub has changed their HTTP headers and has chosen to use lowercase instead of the normal capital first letter.
This results in the header "Location: "(RFC2616:14.30) to have become "location: "
MANY amateur coders have no code in place for case-insensitive parsing because they are not aware!
While I have now found out that the RFC allows for the capitalization to be insensitive, the reasonable thing to do for GitHub is to stick to the text as presented in the RFC and use "Location: "

In my case there are over 20.000 devices out there in the world that depend on my code to update via OTA
HomeACcessoryKid/life-cycle-manager and they cannot anymore.
Of course the self-update of OTA will become smarter, but for the sake of helping the community I really really hope that GitHub will fix this back to the old RFC suggested format. It doesn't hurt them and helps amateurs around the world.
Thanks,
HacK
PS. this has also been posted to the forum for API but I feel there it is a bit out of place...
https://github.community/t5/GitHub-API-Development-and/GitHub-changed-the-capitalisation-of-the-HTTP-headers-and-OTA/m-p/48247 <https://github.community/t5/GitHub-API-Development-and/GitHub-changed-the-capitalisation-of-the-HTTP-headers-and-OTA/m-p/48247>
PPS. I hope this doesn't happen to AWS as well, but for now it is OK
PPPS. There are still 'normal' headers in the reply, just not the top ones anymore
HTTP/1.1 302 Found
date: Sat, 29 Feb 2020 14:47:10 GMT
content-type: text/html; charset=utf-8
server: GitHub.com <http://github.com/>
status: 302 Found
vary: X-PJAX, Accept-Encoding, Accept, X-Requested-With
location: https://github.com/HomeACcessoryKid/life-cycle-manager/releases/tag/1.0.0 <https://github.com/HomeACcessoryKid/life-cycle-manager/releases/tag/1.0.0>
cache-control: no-cache
Age: 0
Set-Cookie: Set-Cookie: logged_in=no; Path=/; Domain=github.com <http://github.com/>; Expires=Mon, 01 Mar 2021 14:47:10 GMT; HttpOnly; Secure
Content-Length: 139
X-GitHub-Request-Id: DF95:2F739:3CBF7FE:57D6ABE:5E5A796E

tonysprenk, 2020-05-13:
@HomeACcessoryKid just got this response from github support. I also sent you a private message on discord. [image]

seritos, 2020-05-13:
@tonysprenk did they ever answer? 
I believe its good news an answer for them... isnt it?

[20 more replies]

closed 2020-05-20 by RavenSystem