RavenSystem/esp-homekit-devices #827: Homekit accessory cannot be found in Home App

paolocrybaby, 2020-03-01:
Hi all, first time I am flashing a device at all (Sonoff S20), I followed the wiki (great job RavenSystem thanks!). 

Working from a Macbook, after erasing the previous firmware, the command to flash the new one doesn't work. (I didn't save the log honestly)

I had to type "esptool.py -p /dev/nameofport --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 haaboot.bin" instead of "esptool.py -p /dev/nameofport --baud 115200 write_flash -fs 1MB -fm dout -ff" - otherwise the process wouldn't start.

After that, led died, doesn't turn on at all. After the successful process, I plugged it again, joined the HAA-XXX network and ok until the JSON config. I used the one in the Wiki for S20, and connected to my wifi 5Ghz.

Still no led, trying to pair it with Home App - the device doesn't show up.

Where is my mistake/s in the process?

Thanks for the support

claire078, 2020-03-01:
fyi
[https://github.com/RavenSystem/esp-homekit-devices/issues/825](https://github.com/RavenSystem/esp-homekit-devices/issues/825)

Bolshoylu, 2020-03-01:
-my wifi 5Ghz.
May be here?

paolocrybaby, 2020-03-01:
Tried again with a Basic R3 and the updated .bin files. It worked! Thanks all!

closed 2020-03-01 by paolocrybaby