RavenSystem/esp-homekit-devices #829: JSON string for Sonoff Dual

Crosss, 2020-03-01:
Does anyone have a working JASON string for Sonoff Dual?
These relays are not controlled via any of GPIO. 
This device has another chip for relay control.
I tryed JSON for 2ch Relay and others, it doesn`t work.
I can`t solve a problem:(

tonysprenk, 2020-03-01:
It could be that the esp is talking to a secondary mcu through serial commands. Serial commands are not (yet) supported by HAA.

xtremedays, 2020-03-04:
this works for my dual 

{"c":{"l":13,"b":[{"g":10,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}]}]}

Crosss, 2020-03-04:
Thank you!
Unfortunately this string does not switch the relays too.
It just blinks with led when i press icon in HomeKit, but relays still dead:(

xtremedays, 2020-03-05:
Potentially a faulty dual then? Mines been working on that JSON since day one...

teymur1988, 2020-03-07:
Hi, 
You might have an old version of sonoff dual, not the R2. In the old version a separate chip is used to control the relays, and that chip gets commands from the ESP chip. HAA works with Sonoff Dual R2. In that version of sonoff the relays are controlled directly by GPIOs of the ESP chip.

Crosss, 2020-03-07:
Hi! You are right. This device is not vR2.
Will have to use another:(

closed 2020-04-06 by stale[bot]