RavenSystem/esp-homekit-devices #833: Sonoff Mini doesn't get into flash mode

paolocrybaby, 2020-03-02:
Hi all,
I am trying to flash a Sonoff Mini (I will not mention how much I sweared about soldering it).
When I plug the FTDI, keeping the button pressed, it doesn't do anything.
It says... connecting and then: "A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header"

The funny thing is that if I plug without pressing anything, the blue led turns on and start blinking (and I am even able to pair it with eWeLink)- perhaps is GND and 3V that just gives power?

Some context: I soldered 4 normal wires to the 4 areas on the back of the board, then I soldered each wire to a pin, and a female Dupont cable connects to each pin. The Dupont cable then goes into the FTDI adapter ending with another female. (I also tried switching R and T)
I tried as well adding the bridge for OTA update (when one is desperate) - nothing

If plugging it in without pressing does turn on the led, can I assume connection is working? How can I understand if wiring is ok?

Thanks for helping!

xtremedays, 2020-03-05:
I have flashed a few, and total understand the frustration!!!! Make sure the soldiers are clean and not shorting our on other pins... they do work!!

paolocrybaby, 2020-03-08:
Thanks stream days for your empathy!! ☺️
I succeeded soldering the DuPont male pin heads directly on the Sonoff board. (So it was my mistake in soldering)

Now I have another question: I’ve flashed both a mini and a basic r3, using the Json in the repository and I cannot recall setup mode (if I understood correctly I should hold the external switch for 8secs)

Anyone else experiencing this issue?

tonysprenk, 2020-03-09:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode

closed 2020-04-08 by stale[bot]

paolocrybaby, 2020-04-23:
Hello Everyone,
sorry to come back here but I still have the same problem, about entering setup mode.

I just flashed a sonoff S26 using this JSON (From The collection):

> {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

(I am not expert of JSONs, but I understood this would go into setup mode if I push and hold the button for 8 seconds)

After pairing with Home, I tried to enter setup mode before unsoldering, in order to check that I can re-setup in case I will change LAN Settings, without reflashing from the beginning... no way I could achieve that.

I tried 4 options (with duponts still soldered and connected to the USB-TTL):
1. Connecting to usb to power it up, pressing 8 secs - doesn’t work
2. Pressing before connecting to USB, holding 8 secs... goes into flash mode, of course - doesn’t work
3. Connecting to power outlet (Without remounting the S26) and press/hold 8 secs - doesn’t work
4. Pressing before connecting to power outlet (Still S26 opened and soldered) and holding 8 secs - doesn’t work

Am I using a wrong JSON?