RavenSystem/esp-homekit-devices #835: Sonoff Mini does not get out of setup mode but accepts Json string

tlagarto, 2020-03-06:
I'm trying to get a simple Json string on a Sonoff mini to work a single button:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}

I've used this string in 2 other sonoff minis and it is working flawlessly (firmware v1.7).
Now I've flashed 2 new ones and they allow me to get in (setup mode) and save the string.
When I get in again, the string and wifi settings are there.
But, strangely enough, they never get out of setup mode (wifi always on) and, thus, do not work.
No light what so ever, no reaction to clicking on button GPIO 0 or using external switch on GPIO 4.

I have flashed both 2 or 3 times and I have no idea what's happening.
I'm using HAA Installer v2.2.2 and firmware v1.8.5.

Any idea?

RavenSystem, 2020-03-06:
Logs, please.

ymyshko, 2020-03-06:
I flashed 2 sonoff mini with HAA 0.9.1 and they work fine. They installed "into the wall" and I don't tryed to update. I udpatet 2 sonoff mini later and got such problem. I receive new sonoff mini from China and I'll try to flash them with HAA 1.8.5. But I don't know when. Solderin need to flash sonoff mini.
I'll test and I'll write results.

tlagarto, 2020-03-06:
> Logs, please.

Must be you. The moment you asked, everything started to work!
Now I can change and save new strings with no issues.
What I did, was saving a new string, request to update OTA and left it on for 30min. When I came back it was working. Maybe it did an update or I was not waiting long enough for it to restart properly...

ymyshko, 2020-04-01:
About Sonoff mini.
Some modules have bad power module. 
Mini takes power from USB-UART when flased and first setup and work good.
But it is not enought power in setup when it takes power from 220V.
Setup mode switch on WiFi AP.
This problem can be solved. But it is hardware problem with very small componetns and dangerous 220V.

closed 2020-05-01 by stale[bot]