RavenSystem/esp-homekit-devices #836: MagicHome Led Strip Controller

stefantitu, 2020-03-07:
Today I flashed one led strip controller from magic home. The flash was a success but after I connect to the WiFi network and save the json config I can not see any new accessory in home app. 
What I noticed is that the WiFi network (HAA-a10df8) keeps appearing among the WiFi networks and if I try to connect again it fails.
I reflashed it for 3-4 times but the same issues appeared.
All my Sonoff were flashed this way and I had no issues connecting them.  
Any ideas?

brymonster, 2020-03-08:
If there’s an issue with that JSON your device will continue booting into setup mode.
What magic home are u using?

stefantitu, 2020-03-08:
[image]

stefantitu, 2020-03-08:
I tried to figure out if there is any mistake in the json code but I don’t find it. I also change the GPIO numbers according to the esp module.

RavenSystem, 2020-03-08:
Logs, please.

stefantitu, 2020-03-08:
> Logs, please.
I not an expert and I don't know where to find the logs, but if you tell me then I paste them here.

tonysprenk, 2020-03-08:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-Instructions

stefantitu, 2020-03-08:
[screenlog.txt](https://github.com/RavenSystem/esp-homekit-devices/files/4303140/screenlog.txt)
this is the log. thank you

RavenSystem, 2020-03-08:
You are using an old version of haaboot.bin.

stefantitu, 2020-03-08:
Where do I find a newer version?

closed 2020-04-08 by stale[bot]